bambulab/BambuStudio #4213: Scaling a part in X scales it in Z after rotating (version 1.9)

emberprototypes, 2024-06-04:
### Bambu Studio Version

1.9

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

X1C

### How to reproduce

See my Twitter post here: https://x.com/emberprototypes/status/1797859883183640670

- Import a part into Bambu studio
- Rotate it by laying a different face flat
- Unclick "uniform scaling" and try to scale in X and Y
- Version 1.9 will scale incorrectly

[image]
[image]

If you revert to 1.8.4 it scales as expected. I believe all previous versions worked this way.
This is important because I scale parts manually to accommodate for XY shrinkage for materials like ABS and PC.
See this issue for the need for shrinkage compensation: https://github.com/bambulab/BambuStudio/issues/2769

[image]

### Actual results

Part is scaled incorrectly.

### Expected results

Part should global scale in X and Y appropriately.

### Project file & Debug log uploads

N/A - should work with any STL.

### Checklist of files to include

- [X] Log file
- [X] Project file

Haidiye00, 2024-06-05:
[image]
Hello, our scaling tool defaults to using the local coordinate system, which is the xyz direction of the local coordinate system. We will launch a scaling tool for the world coordinate system later, please stay tuned.

emberprototypes, 2024-06-05:
This is not how it worked in previous versions though - if you try it on 1.8.4 and below, scaling values always applied to the world coordinate system, not the local coordinate system of the part.

Haidiye00, 2024-06-05:
It used to be wrong

emberprototypes, 2024-06-05:
I see. Was this in the release notes? This caused issues on my end because I was using the scaling function to compensate for shrinkage and because its behavior suddenly changed, I encountered lots of parts out of tolerance.

Please add a global scaling function in the future (or more importantly...implement the filament type shrinkage compensation).

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00