bambulab/BambuStudio #4214: Bad Spanish translation

Twikzer, 2024-06-04:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1. Click on prepare
2. click on any fillament to open the dropdown
3. click on add/remove fillaments

### Actual results

Some text is in italian

### Expected results

All the text is in spanish

### Project file & Debug log uploads

[debug_network_Tue_Jun_04_23_31_06.log.zip](https://github.com/user-attachments/files/15570874/debug_network_Tue_Jun_04_23_31_06.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Twikzer, 2024-06-04:
[image]
[image]

StoneLiBambu, 2024-06-28:
hi @Twikzer 
I will try to reproduce it and solve it.

Stone Li
Thanks!

Twikzer, 2025-03-26:
still hapens on 2.0.0.95