bambulab/BambuStudio #4215: Mirror the enhancements to Cross Hatch made in Orca Slicer v2.1.0 Beta

sheehand, 2024-06-04:
I used the new cross hatch infill on a bowl, but the infill ended up leaving left large gaps next to some of the walls, which led to deformation where the walls pulled inward to the print. I didn't notice it until after the print was complete, otherwise I would have chosen gyroid or some other infill.

Here is a screen capture of what I am referring to where there is a gap between the infill and the wall of the print:
[image]

It seems as if I am not the only one who noticed this because the Orca Slicer team have enhanced the Cross Hatch and their modifications look like it addresses what I ran into. Here is their pull request on the issue:
[optimize cross hatch infill to enhance strength #5495](https://github.com/SoftFever/OrcaSlicer/pull/5495)
Reference [OrcaSlicer V2.1.0 Beta Release](https://github.com/SoftFever/OrcaSlicer/releases/tag/v2.1.0-beta)

Hopefully since the Orca Slicer team already did a lot of the heavy lifting the changes can be pack ported to Bambu Studio.

dragon-eyes-bbl, 2024-06-05:
Thanks for mentioning @sheehand , we noticed the enhancement. The enhancement will be implemented in the next version after it passes our testing.

SaltWei, 2024-06-05:
4 lines code, and very easy to port. @dragon-eyes-bbl Please handle it.

@sheehand Thank you very much for suggestion!

sheehand, 2024-06-05:
Thank you all!

closed 2024-06-26 by SaltWei