bambulab/BambuStudio #4218: Inconsistent layer lines with multi-support filament

SinanAkkoyun, 2024-06-05:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab A1 + AMS Lite (got it 5 days ago)

### How to reproduce

0.4mm nozzle, 0.16mm optimal (any layer height will reproduce it though)

[Use completely stock settings]

1. Load model with overhangs, Bambu PLA (silver for best reproducibility)
2. Enable support with PETG support interface
3. Look at `layer times` prior to printing
4. Print
5. Observe layer time inconsistencies exactly lining up with print artifacts

### Actual results

Layer artifacts are NOT shifts, they are thicker extrusion all around the model. As if the nozzle is still

[image]
[image]
[image]

### Expected results

Given that wall speeds are all the same, the print should look flawless and have no layer inconsistencies.

### Project file & Debug log uploads

[test_model.zip](https://github.com/user-attachments/files/15586543/test_model.zip)
Keep in mind, this should happen with any model.

### Checklist of files to include

- [ ] Log file
- [X] Project file

SinanAkkoyun, 2024-06-05:
I just ran a print with `Bambu Support for PLA` and the lines were much better, but still fundamentally there:

[image]

The supportsl base looks the same artifact wise.

What does this mean? I need to get rid of all faulty layer extrusions whilst having good support

QingZhangBambu, 2024-06-06:
@SinanAkkoyun I will make the layer time smoother to improve the surface.