bambulab/BambuStudio #4220: Windows on ARM (ARM64/AArch64) Support

hifihedgehog, 2024-06-05:
**Is your feature request related to a problem? Please describe.**
I have an ARM-based Windows machine and would like to use it with Bambu Studio for PC-directed printing. I can run the x86-64 build, but at a tremendous penalty to both performance and battery life. A native ARM build would resolve these issues.

**Describe the solution you'd like**
I'd like an ARM native build of Bambu Studio for Windows. This would put ARM-based hardware for Windows on par with Apple ARM hardware.

**Describe alternatives you've considered**
Running an x86-64 build on ARM is possible, but at a major performance and battery life penalty.

**Additional context**
Qualcomm Snapdragon X series processors are taking the market by storm this month with historic crossroads and massive push from all major PC makers. Microsoft is giving priority development and support to ARM and relegating x86 to second-class citizen.

fijsh, 2024-07-05:
Yes, this would definitely be great to have!

myshophosting, 2024-07-12:
I also now run an ARM laptop with Windows 11.. An ARM native version would be awsome.

goat77, 2024-08-18:
I would also like to see this.

BillsWorkbench, 2024-08-19:
Yes, please; the intel version is "quick exiting" on my Arm64 Notebook. (snapdragon elite)

andrew-c0, 2024-10-27:
One up from me as well. Would be great to have native support.

danielcherubini, 2024-11-13:
Yes please

mhertel, 2024-12-10:
I would like to see it as well!

adachi86, 2024-12-28:
+1 (Surface Laptop 7 w/ Snapdragon X Elite)

pleite, 2025-01-12:
+1 really slow using preview on Surface Pro X (first edition).

dirtdiver99, 2025-01-22:
+1, this would be amazing.

spbrogan, 2025-02-16:
Agree.  Please provide native application for this environment.

RyanEwen, 2025-02-25:
Brand new Surface Laptop 7 here and Bambu Studio closes when launched (seems related o network plugin).

Oddly, Orca Slicer works fine

RyanEwen, 2025-02-25:
Hmm apparently 1.10.1 works... it's just 1.10.2 that crashes for me.

adachi86, 2025-03-01:
> Hmm apparently 1.10.1 works... it's just 1.10.2 that crashes for me.

Same

ludovandenbrink, 2025-03-13:
+1 would love to see this.

egeerardyn, 2025-04-17:
+1 here

konashevich, 2025-08-31:
AARCH64 need with compatible network plugin! Dear developers, stop ignoring the community! Either issue opens source plugin (accountless for example if you don't to disclose this part of the code, or something) or add support to arm64/aarch64. People ask you for many years! How long will you ignoring it?

generally-ready, 2026-08-18:
+1 (not that it seems to matter, since this PF has been open since 2024...)

RyanEwen, 2026-08-18:
I added this a month ago - it's just sitting there waiting to be looked at, but not being looked at: https://github.com/bambulab/BambuStudio/pull/11599

generally-ready, 2026-08-18:
> I added this a month ago - it's just sitting there waiting to be looked at, but not being looked at: [#11599](https://github.com/bambulab/BambuStudio/pull/11599)

Amazing, @RyanEwen .  By chance, does your PR also address [#11045](https://github.com/bambulab/BambuStudio/issues/11045)?  Not critical, but bloody annoying...

RyanEwen, 2026-08-18:
> Amazing, [@RyanEwen](https://github.com/RyanEwen) . By chance, does your PR also address [#11045](https://github.com/bambulab/BambuStudio/issues/11045)? Not critical, but bloody annoying...

I don't think so but I haven't really compared to know. I've definitely seen that bug, but didn't think it could be arctitecture-specific. 

To be honest, I have been rebuilding as much of the Bambu Studio UI into (shameless plug) [PrintStream](https://printstream.app) and avoiding the native application as much as I possibly can.