bambulab/BambuStudio #4225: Add G-code: Right clicking layer selection bar causes layer to change

SimEyeSee, 2024-06-06:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 14.4.1

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. select layer in layer selection bar
2. right click to bring up pop up box with option to add custom G-code

### Actual results

The layer jumps and selects the wrong layer to add the G-code. The higher to model, the more difficult it is to select the layer I want.

### Expected results

In addition to selecting a layer with the mouse, after the mouse has selected the layer bar, we can currently click the up and down arrows to move between layers. It would be ideal if you could add a keyboard shortcut so that after clicking the up or down arrow to get to a specific layer, I can click the shortcut key and this will bring the pop up box for that specific layer with the option to add G-code.  This approach will make it so much easier to add G-code to a specific layer.

### Project file & Debug log uploads

[FanFrame v167 - ClipBender2.3mf.zip](https://github.com/user-attachments/files/15601005/FanFrame.v167.-.ClipBender2.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

LiZ-Li-BBL, 2024-06-12:
This issue may occur when using a touchpad, as the a left-click is detected first, which cause the layer to change. We will optimize this in a future version.

SimEyeSee, 2024-06-12:
Yes, I was using a touchpad. Thanks for looking into fixing the issue.