bambulab/BambuStudio #4227: Unreliable GCode generation

y-nk, 2024-06-06:
### Bambu Studio Version

latest

### Where is the application from?

Bambu Lab Official website

### OS version

macOS latest

### Printer

P1P

### How to reproduce

1. slice a model
2. preview the gcode
3. slice again (turn Z to 1, then 0 to re-enable the button)
4. preview the gcode

### Actual results

it is not the same

provided recording:

1st slice shows a "counter clock wise" progression
2nd slice shows a "clock wise" progression

[image]

### Expected results

it should be the same

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-06-17:
Hi, the path sequence is decided by an algorithm that finds the shortest path.  The shapes of the bottom surfaces are the same, and the connection directions of the paths can be random. Counterclockwise or clockwise does matter as long as the time is the same.

y-nk, 2024-06-19:
> the connection directions of the paths can be random

does that mean, in the event of mass production of a print, that there's no way to be sure that the print sequence will be the same? or we should save the gcode and re-run it at will since slicing produces a new gcode?

QingZhangBambu, 2024-06-19:
> > the connection directions of the paths can be random
> 
> does that mean, in the event of mass production of a print, that there's no way to be sure that the print sequence will be the same? or we should save the gcode and re-run it at will since slicing produces a new gcode?

yep.  normally it will affect the print result quality. If you want ensure the sequence be the same, save gcode file.

y-nk, 2024-06-20:
@QingZhangBambu is there a feature to decide/influence the randomness of path? i would like that the little squares in my gif example be printed from left → right and top → down to minimize printer movements.