bambulab/BambuStudio #4228: Crash after sending print over network

pbronez, 2024-06-06:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 14.5 (23F79)

### Additional system information

|Key|Value|
|------|------|
|Model Name|Mac Studio|
|Model Identifier|Mac14,13|
|Model Number|Z17Z000MELL/A|
|Chip|Apple M2 Max|
|Total Number of Cores|12 (8 performance and 4 efficiency)|
|Memory|64 GB|
|System Firmware Version|10151.121.1|
|OS Loader Version|10151.121.1|

### Printer

Bambu Lab X1C

### How to reproduce

1. Slice a plate
2. Send plate to printer

### Actual results

Crash

### Expected results

Not crash

### Project file & Debug log uploads

**Project File**
[Crashing Project.3mf.zip](https://github.com/user-attachments/files/15616535/Crashing.Project.3mf.zip)

**Bambu Debug Log**
[debug_Thu_Jun_06_07_14_58_13020.log.txt](https://github.com/user-attachments/files/15617271/debug_Thu_Jun_06_07_14_58_13020.log.txt)

**Apple Crash Report**
[BambuStudio-2024-06-06-080448.ips.txt](https://github.com/user-attachments/files/15617874/BambuStudio-2024-06-06-080448.ips.txt)

I've encountered this error several times. If needed, I can probably dig up more Logs and Crash Reports.

### Checklist of files to include

- [X] Log file
- [X] Project file

walterwongbbl, 2024-06-14:
Hi. @pbronez 
I'm very sorry for replying so late. If you could provide a crash reopt, we would greatly appreciate it.

pbronez, 2024-06-14:
I uploaded an apple crash report in the initial post. Do you need more of the same thing or do you need something different?