bambulab/BambuStudio #4232: Error while loading shared libraries: libwebkit2gtk-4.0.so.37

mfacton, 2024-06-07:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab github releases

### OS version

Ubuntu 24.04 LTS

### Additional system information

Worked on Ubuntu 22.04, but not 24.04, even with downgrading libwebkit2gtk from 4.1 to 4.0

### Printer

Any

### How to reproduce

sudo apt-get update
sudo apt-get upgrade
sudo apt-get install libwebkit2gtk-4.0
sudo chmod +x bambu-studio.AppImage
./bambu-studio.AppImage

### Actual results

/tmp/.mount_bambu-wQgUx9/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Expected results

The software should open

### Project file & Debug log uploads

Does not make log (doesn't boot)

### Checklist of files to include

- [X] Log file
- [ ] Project file

mfacton, 2024-06-08:
I just had to make symbolic links for the libwebkit2gtk-4.1 and libjavascriptcoregtk-4.1 dependencies and it works now

closed 2024-06-08 by mfacton

EggzBacon7, 2025-02-11:
sudo apt-get install libwebkit2gtk-4.0 

isso realmente funcionou !!!

darbdarb, 2025-03-14:
same issue on fedora

```
./Bambu_Studio_linux_fedora-v01.10.02.76.AppImage 
/tmp/.mount_Bambu_l99QjE/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```

MechEng70, 2026-01-05:
Seems like the latest fedora AppImage is still looking for the webkit2gtk4.0.

./Bambu_Studio_linux_fedora-v02.04.00.70.AppImage 
/tmp/.mount_Bambu_MOMAbj/bin/bambu-studio: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

Flatpak is slow and zoomed in.  Frustrating.

Fonant, 2026-10-01:
I'm getting this too, Fedora 44.

Faffing around with symlinks to old library versions gets Studio to open, but then I have multiple runaway WebKitWebProcess processes eating up 100%  CPU each.

New report, including logs:  #12433