bambulab/BambuStudio #4233: Varying initial flushing volumes based on first filament color used on first layer

hcglitte, 2024-06-07:
Since the printer likely will have, no or at least poor knowledge, of the last filament type and/or color used in the previous print (the very last layer etc), color bleeding on the initial layer of the next print may experience color bleeding.

An example is if PLA was used on the last print, and then PETG is used on the second print.
PLA should be flushed at great amounts before PETG.

An other example: Last color used in previous print was black, and the first color used in the next print is white. Currently the first layer will have some grey areas (where the first print lines are).

Now there is an equal initial flushing volume regardless og the first color (or material) used.

This flushing volume could instead be set to have the "worst case" amount required in order to ensure a pure color on the first layer.

If the printer could store in memory which was the last color/material type used, the initial flushing volume required for next print could be "smart".

luder888, 2025-01-04:
Ditto on this. I can't count how many of my white prints are ruined because previously I had printed a dark colored model... this needs to be a basic feature.

TshineZheng, 2025-02-08:
This improvement is urgently needed. When printing a PETG model immediately after using PLA, it causes the bottom of the PETG model to become brittle.

unCleanCode, 2025-04-08:
It's a shame there is still no solution to it. For now I solved it by changing start GCode.

From:
G1 E50 F200  ; MAX: Prevent dark color bleeding from the previous print

to:
G1 E100 F200  ; Prevent dark color bleeding from the previous print

this would change flushing from 50mm to 100mm. Unfortunately for all prints, but that's not huge amount

BenGeba, 2026-11-04:
> 4\. It's a shame there is still no solution to it. For now I solved it by changing start GCode.
>         From:
>         G1 E50 F200
>         to:
>         G1 E100 F200  ; Prevent dark color bleeding from the previous print
>         this would change flushing from 50mm to 100mm. Unfortunately for all prints, but that's not huge amount

Yeah, that's exactly how I do it, too. It's a shame there's still no way for the printer to remember the last filament color and use that to calculate how much flushing is needed.