bambulab/BambuStudio #4237: Upload log files through Bambu Handy

texasmadevape, 2024-06-08:
### Bambu Studio Version

01.05.02.00

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu P1S

### How to reproduce

Log files from the P1 series printer can be uploaded via Bambu Handy over the network, similar to the X1 series, with an upload speed of around 150 kbps under optimal network conditions.

When uploading logs, you have the option to select a specific time frame, allowing you to reduce the volume of logs transmitted and shorten the upload duration.

To upload files, go to Me → Support Tickets → Choose the ticket for log upload and proceed with the operation.

### Actual results

not there

### Expected results

to upload log files for support ticket

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15748422/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-06-11:
do you mean you expect to upload bambustudio log like printer logs?
@texasmadevape

texasmadevape, 2024-06-11:
I was trying to create a ticket because I was having issues and still am
trying to get my printer to start working again. When filling out a ticket
it asked for me to upload logs and that it was needed to submit the ticket.
It said I would be able to upload them remotely from my P1S through the
Bambu Handy. I followed the steps up to a point and then there was no way
to finish it or the help guide is outdated,