bambulab/BambuStudio #4238: 3D Connection Menu

Jgelwix, 2024-06-08:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

14.5 (23F79)

### Printer

X1

### How to reproduce

Command+M for 3D Connection Settings - Never been able to use this as this shortcut is the Mac global shortcut for minimize. It would be great to see this fixed, or even better, the ability to customize/change keyboard shortcuts. Thanks in advance!

### Actual results

Window is minimized

### Expected results

3D Connection Menu

### Project file & Debug log uploads

N/A - This is a keyboard shortcut issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-06-11:
[image]
Can you try this shortcut button? Thank you

Jgelwix, 2024-06-11:
Unfortunately, this also minimizes the window.

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00