bambulab/BambuStudio #4239: crashes when slicing

chimeraproduction, 2024-06-08:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

cpu intel core i9-14900k
ram 64gb
gpu amd radeon rx 6950 xt
vram 16368

### Printer

bambu lab x1 carbon

### How to reproduce

dont know how to reproduce

### Actual results

software would crash when slicing prints

### Expected results

software would crash when slicing prints

### Project file & Debug log uploads

[crash_Sat_Jun_08_14_35_40_0_done.log](https://github.com/user-attachments/files/15749616/crash_Sat_Jun_08_14_35_40_0_done.log)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

SpunkyTaz, 2024-06-09:
I am having the same issue on my P1S. It only seems to be happening with a larger print I am trying to do.

QingZhangBambu, 2024-06-11:
Could you try these methods? and whether it has functions such as automatic overclocking enabled? 
If these methods do not work. could you upload your dump file and 3mf file?

chimeraproduction, 2024-06-15:
> Could you try these methods? and whether it has functions such as automatic overclocking enabled? If these methods do not work. could you upload your dump file and 3mf file?

what methods? and the crash file is in my post