bambulab/BambuStudio #4240: Device tab control section not fitting screen properly.

schurter-richard, 2024-06-08:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Lenovo X1 Yoga Gen 7

### Printer

X1C

### How to reproduce

1) Open Bambu studio.
2) Click on device tab
3) Observe right side and bottom of control section cut off. (I don't think it is intentional the bottom of the control panel and the bottom of the printing progress panel do not line up.)
To use control section, you have to carefully use the side/bottom scroll wheels to access, and it gets laggy when you do that. There is not any way to zoom out to fit everything on the screen.

### Actual results

[image]
The scrolling causes the app to get very laggy as well.

### Expected results

The tab should neatly fit all the control panel within the screen so scrolling is not necessary, and so that it lines up with the rest of the page.

### Project file & Debug log uploads

This is not related to any specific project, so I am not going to upload one.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

schurter-richard, 2024-06-08:
Note: My other computer with a larger screen does the opposite; it appears that the camera and printing progress panels scale, but the control panel does not, so the control panel is smaller than it should be.
[image]

walterwongbbl, 2024-06-14:
Hi. @schurter-richard 
This is indeed a problem, and we will adjust this page.

GT1man, 2024-12-14:
This looks like it is still an issue. Is there any updates?

[image]

DouglasMorency, 2026-08-29:
On my screen, which is an RCA LCD TV, with HDMI Input Running Windows 11. I cannot hide the left side of the screen, like you can in the prepare and preview tabs, and therefore, I cannot see the entire right part of the screen as shown above.