bambulab/BambuStudio #4241: Split to parts slicing issue

schurter-richard, 2024-06-09:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Lenovo Thinkpad X1 Yoga Gen 7

### Printer

X1C

### How to reproduce

-Have a part with two pieces that need to be printed, in my case it is a single layer of a different color in the middle of a print.
-After importing to Bambu Studio, split to parts. Then make the single layer a different filament.
-Slice the part
-Check the layer that is a different color

### Actual results

Instead of getting a single uniform layer, it takes on the pattern of the layers above/below it. This causes overhangs to fail, and the print to be ugly. One way to get around this if the first layer is the one with a different color is to split to object instead, but this does not work if the second color is not on the first layer.
[image]
[image]

 [image]

### Expected results

I would expect the layer to be printed as if it were the bottom/top layer. If you look at the image where I split it to objects it works great. The problem is I need the secondary color to be in the middle of the part, which does not work when splitting to object.
[image]

### Project file & Debug log uploads

[debug_Sat_Jun_08_13_52_57_35660.log.zip](https://github.com/user-attachments/files/15749926/debug_Sat_Jun_08_13_52_57_35660.log.zip)
[Bambu Studio Split to part slicing issue.zip](https://github.com/user-attachments/files/15749929/Bambu.Studio.Split.to.part.slicing.issue.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

zhimin-zeng-bambulab, 2024-06-13:
[image]

@schurter-richard  Does your problem mean that the print quality in these areas is poor?

schurter-richard, 2024-06-13:
> ![image](https://private-user-images.githubusercontent.com/120697589/339191562-cec1928f-0cd2-43e9-9aa5-059a12defca1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Yes, poor quality like shown in your image. And it also will leave gaps between extrusion lines depending on the model being printed.