bambulab/BambuStudio #4242: 打开软件显示Microsoft Edge无法读取和写入其数据目录，主页显示空白，无配置界面

physicsexpert, 2024-06-09:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Additional system information

CPU:AMD Ryzen 7 5800H with Radeon Graphics
显卡：NVIDIA GeForce GTX 1650
内存：16G
硬盘：512G

### Printer

A1 mini

### How to reproduce

1.双击软件图标
2.按ESC键点
3.点击主页

### Actual results

[image]
[image]
[image]

### Expected results

双击软件图标后应该正常打开配置界面和主页，可以进行登录，可以配置选项

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15750445/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

bambu123, 2024-06-11:
看起来像是目录权限问题，试试手动创建这个目录，看看能不能创建。

physicsexpert, 2024-06-11:
已经解决了
问题原因：C:\Users\<用户文件夹>最终用户读写权限权限未打开
解决方法如下：检查C:\Users\<用户文件夹>的权限，以确保当前用户帐户具有读写权限（可能需要为“所有人”授予该权限）

bambu123, 2024-06-11:
好的，感谢反馈！

closed 2024-06-11 by bambu123