bambulab/BambuStudio #4243: Modified Layers affecting Top and Bottom Layers

Vezide, 2024-06-09:
### Bambu Studio Version

V1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambu Lab P1S

### How to reproduce

1. Open a Model
2.  Set to 0.16mm layer height
3. Add a modified Layer
4. make modified layer 0.3mm height
5. place modified layer on the very top or bottom just clipping the first few layers
6. Set the modified layer to : 0 top layers 0 bottom layers, and pick the infill pattern you want (eg. 25% Honeycomb)
7. Slice

### Actual results

There's holes all the way through the part, instead of the top and bottom layers kicking in originally where they would, it instead immediately turns to the parts original infill and leaves massive holes.

### Expected results

In Ever past version of Bambu Slicer, The modified layer only affects the layers its interacting with, and the rest of the part continues as it always had. So you should see clean pattern followed by several solid layers then the infill, rather than Clean pattern and immediately infill.

### Project file & Debug log uploads

[Top Glass Riser Final Assembly.zip](https://github.com/user-attachments/files/15750468/Top.Glass.Riser.Final.Assembly.zip)
This is an example of a project I made where you can see the issue, i also attached the link to the original maker lab URL where you can see images of what it should've looked like
https://makerworld.com/en/models/194103#profileId-223721

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-06-11:
the thinckness of modifier is 0.3mm, the z of center point is 0.25.  So the height range the modifier covered is from 0.1 to 0.4. The first layer height is 0.2mm. first layer will be affected by modifier.

Vezide, 2024-06-11:
Yeah that makes sense, i want the modifier to affect the first layer, even the second layer too, however i dont want the modifier to affect the 5th layer or the 10th, if I set bottom shell layers to 10, it doesnt make a difference, it removes all shells and only does infill, i want to keep the shell layers, and not have them replaced my infill, the only solution ive found is to have 0 infill, and have only shell layers all the way through

Vezide, 2024-06-11:
Yeah that makes sense, that's not the issue I'm having. I want the modifier to affect the first layer, even the second layer too, I want the modifier to affect the layers that the modifier is covering. However, I don't want the modifier to affect the 5th layer or the 10th, when it's set to only the 4th.  If I set bottom shell layers to 10, it doesn't make a difference, it removes all shells and only does infill, I want to keep the shell layers, and not have them replaced my infill, the only solution I've found is to have 0 infill and have only shell layers all the way through. And this has only changed after the first update, I've been using this method for over a year to make several different designs, only after the new update does a modifier affect the whole part.

QingZhangBambu, 2024-06-11:
[image]
do you mean you want this? like version 1.8.4

QingZhangBambu, 2024-06-11:
[image]
do you mean it should have bridge path at 0.6 height?

QingZhangBambu, 2024-06-11:
I get you, it affects the outlook.

Vezide, 2024-06-11:
Yes 100%, im not sure if that screen shot is from an earlier version, but that's exactly what it should look like. I'm unable to get a bridge and layers upon it, is there some setting I'm missing for it? I even updated today to the even new version 1.9.2.57 but sadly I still have the same problem I was having.

XunZhangBambu, 2024-06-14:
@Vezide Currently turn on the interface shell option and then you can get what you want.  
[image]

Vezide, 2024-06-14:
Hey @XunZhangBambu thanks for the help, where can I find that setting? when I search for interface shell nothing comes up. I'm able to find Interface layers, and interface Pattern. can you show a screen shot with the setting? 
[image]

XunZhangBambu, 2024-06-14:
@Vezide  The setting is on develop mode

Vezide, 2024-07-06:
@QingZhangBambu @XunZhangBambu , Thanks for the help, it's been a few weeks now and I'm wondering, when will this setting become available to the public? rather than being in beta? right now, this setting may fix the issue, but the issue didn't exist until the recent updates, and I'm guessing it was in order to help people having this issue: 
https://github.com/bambulab/BambuStudio/issues/2906
However, their issue isn't fixed, and neither is the problem I've been having.  I'm unable to make changes in the Beta and upload to maker world, and people cannot download and use it. Do you have a rough idea of when this will be implemented? or fixed back to the original way it used to be? Currently I've been using orca slicer in the meantime to print a lot of things, because Bambu isn't able to handle it