bambulab/BambuStudio #4247: Reload custom negative part fails with error

Regis-Frollo, 2024-06-09:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

Bambulab P1S

### How to reproduce

Load attached project
Chose kegel.step
Right click on that part
Chose Reload from Disk

### Actual results

Get error message 'Error during reload' (see file attached)
[image]

### Expected results

Reload of the file / no error

### Project file & Debug log uploads

[3d-Druck.zip](https://github.com/user-attachments/files/15751704/3d-Druck.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

MackBambu, 2024-06-14:
Are you sure the STEP file didn't change the path ? You need to make sure it's still in “C:\Users\FOL\Nextcloud\3d-Druck\Kegel.step”

Regis-Frollo, 2024-06-14:
Yes I am sure. I have loaded Kegel.step and immediately try to reload it and it fails. But if it is not a negativ part reload works well.

Regis-Frollo, 2024-06-14:
Not like that. You import as general part. I add negativ part by right click on Homi_Hand_Plow_Left.step select add negative part and then select Kegel.step.

MackBambu, 2024-06-18:
This is an upstream issue
https://github.com/prusa3d/PrusaSlicer/issues/11332