bambulab/BambuStudio #4250: Layer pause is shifted by two layers

jacobtc, 2024-06-09:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

Bambu Lab X1 Carbon

### How to reproduce

Add a pause to a model. The gcode produced will pause the print at layer 27, whereas it is selected as layer 25 in the slicer.

### Actual results

[image]
[image]

The print will pause at the wrong layer.

### Expected results

The print should pause at the start of the selected layer.

### Project file & Debug log uploads

[Gcode pause.zip](https://github.com/user-attachments/files/15754190/Gcode.pause.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

freedimension, 2024-08-28:
Am experiencing the same problem. Had a layer paused but it didn't stop at the start of it but on a later one (could have been one or two layers in, couldn't tell exactly). Ruined that print. With another print of the exact same object it however did what it should; stopped right at the correct height and I could insert my objects. The only difference I could tell is, that the print (that is failing to do so) also has an AMS color change on a later level programmed. Perhaps it then miscounts somehow!?

XunZhangBambu, 2024-09-06:
@jacobtc I did not find the offset for "pause" in the 3MF you provided. Is variable layer height enabled in your project? I suspect the issue may be due to floating-point precision.

jacobtc, 2024-09-06:
> @jacobtc I did not find the offset for "pause" in the 3MF you provided. Is variable layer height enabled in your project? I suspect the issue may be due to floating-point precision.

The problem was fixed with a newer version of Bambu Studio

freedimension, 2024-10-01:
@XunZhangBambu If I remember it correctly, it might have happened right after I reduced my print time by using variable layer height. Floating-point precision is a very reasonable guess, as it would also explain the randomness of it happening.

XunZhangBambu, 2024-11-01:
@freedimension  Hi,in beta version, we have solved the issue. You can try with beta version studio.