bambulab/BambuStudio #4251: Adjusting position of connected parts doesn’t work.

SxeyChick, 2024-06-09:
### Bambu Studio Version

1.9.1

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

P1S

### How to reproduce

Import a part connected to a part on the plate. Then try to move it by inputting numbers. 

https://github.com/bambulab/BambuStudio/assets/115029650/64554e31-d7ac-4eb3-905a-086588d888e6

### Actual results

It’ll move around completely randomly. Same thing happens if you try to rotate it, it rotates at random intervals. 

### Expected results

It should move to the location you input. 

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Haidiye00, 2024-06-11:
We are already upgrading the translation tool internally and in beta testing. Stay tuned

SxeyChick, 2024-06-11:
> We are already upgrading the translation tool internally and in beta testing. Stay tuned

Cool. Though you guys just released an update and didn't fix this. Very disappointing. It's literally unusable. And that sounds like in exaggerating, but no. If you can't define the position of parts, the entire feature of combining two parts it's completely unusable.

Haidiye00, 2024-06-12:
[image]
Sorry, the external version is not yet available, but the internal version has indeed been developed and is being tested. Version 1.10 is definitely usable. Stay tuned.

SxeyChick, 2024-06-15:
> [image]
> 
> Sorry, the external version is not yet available, but the internal version has indeed been developed and is being tested. Version 1.10 is definitely usable. Stay tuned.

I'll just continue to keep most my projects on hold...

Haidiye00, 2024-10-16:
The new version 1.10beta has been released. Please download and use~

closed 2024-10-16 by Haidiye00