bambulab/BambuStudio #4253: Fuzzy skin prints same noise profile on consecutive layers

SeaRyanC, 2024-06-10:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Intel i7-12700K

### Printer

Bambu X1 Carbon

### How to reproduce

* Load the attached STL file
* Choose Advanced -> Fuzzy Skin -> Contour Only
* Slice plate
* Pan the camera around a bit

### Actual results

Here's a view in the slicer showing the same noise pattern persisting over multiple layers
[image]

This isn't a software artifact; the actual print also has this defect. This is a different slice that shows the same defect. Note that the repeated section occurs in a different section of the part.
[image]

### Expected results

There should not be observable repetition of the fuzzy noise pattern

### Project file & Debug log uploads

Attached
[fuzz.zip](https://github.com/user-attachments/files/15757322/fuzz.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SeaRyanC, 2024-06-10:
This can be repro'd with a regular cube primitive. Scale it up vertically and apply fuzzy skin. It takes a few tries and can hard to spot, but there are definitely sometimes repeated layer sequences

Here you can see three identical layers near the top, separated by a bit. Lower, the same _sequence_ of repetitions occurs three times in the same cube

[image]

QingZhangBambu, 2024-06-17:
@SeaRyanC Great, you do a good job!

QingZhangBambu, 2024-06-24:
@SeaRyanC Hi, it had been merged, and will be release on the next version. nice work!

ttt888zzz, 2024-08-27:
HI，[SeaRyanC](https://github.com/SeaRyanC)  Please try updating the latest version 1.9.4.
https://bambulab.com/en/download/studio

closed 2024-08-27 by QingZhangBambu

HDHristov, 2025-02-01:
I'm having the same repeating patterns issue with Bambu Studio 1.10.2.64 when using arachne.
Works ok when using classic wall generator.