bambulab/BambuStudio #4255: I logout google account, now I want to relogin google account, the bambo studio exit while the edge browser google login page appeared....

netenken, 2024-06-10:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 

### Additional system information

i5-14600K
32GB RAM

### Printer

A1

### How to reproduce

1. I logout google account
2. Relogin google account
3. the bambo studio exit while the edge browser google login page appeared....

### Actual results

i5-14600K
32GB

### Expected results

Bambo studio exit..

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15776805/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-06-11:
[image]
[image]

I can login successfully using google acount

could you give a gif or video to help us understand your issue?

netenken, 2024-06-12:
I just try to remove or restrict all the extention , But It's still raised...