bambulab/BambuStudio #4257: 切片后外墙生成错误

muyir123, 2024-06-11:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

WIN11

### Printer

A1

### How to reproduce

使用最新版本切片，切片后在外墙没有的地方生成了多余的线条，关闭外墙识别后正常，应该是外墙识别的bug，现在只能降回低版本使用。

### Actual results

外墙没有的地方生成了两层多余线条
[image]
[image]

### Expected results

生成错误的路径，使用可变外墙后会好一点，但是还会突出，关闭外墙识别后正常。

### Project file & Debug log uploads

[项目文件.zip](https://github.com/user-attachments/files/15789556/default.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

muyir123, 2024-06-11:
从1.9测试版本开始出现的bug，降低到1.8.4.51后bug没了，可以查看下到1.9版本时外墙识别部分优化的代码

QingZhangBambu, 2024-06-12:
你好，1.9版本为提高运行速度，和速度平滑更改了外墙识别的部分功能，是一个第三方库的计算错误。略微旋转一个小角度之后就可以获得正确的计算结构。这个bug会在以后进行修复。

muyir123, 2024-06-12:
> 你好，1.9版本为提高运行速度，和速度平滑更改了外墙识别的部分功能，是一个第三方库的计算错误。略微旋转一个小角度之后就可以获得正确的计算结构。这个bug会在以后进行修复。
您好，这个模型实际上是80*80的尺寸，A1满板刚好9个，如果旋转45度后将只能打4个，会很麻烦，从1.9第一个测试版开始出现bug，之后每次发布都第一时间更新，到正式版，再到第二个正式版还没有修复，非常失望，为了一次打9个，现在还在用着1.8.4.51版本，少了很多功能，可以询问一下大概什么时候修复吗？谢谢！

QingZhangBambu, 2024-06-12:
不需要这么大，z方向旋转0.1°即可，此时对于放置影响不大。
修复时间暂时未定，这个第三方库的问题较为难处理，我会尽力加快处理。

muyir123, 2024-06-12:
> 不需要这么大，z方向旋转0.1°即可，此时对于放置影响不大。 修复时间暂时未定，这个第三方库的问题较为难处理，我会尽力加快处理。

哦哦，好的，谢谢，我试试，麻烦尽快哈

muyir123, 2024-09-19:
> 不需要这么大，z方向旋转0.1°即可，此时对于放置影响不大。 修复时间暂时未定，这个第三方库的问题较为难处理，我会尽力加快处理。

从反馈到现在三个月了，更新了好几个版本，现在最新的版本也还没有修复这个BUG，后续还会修复吗

QingZhangBambu, 2024-09-20:
@muyir123 会的，修复已经在做内部数据集验证。

muyir123, 2024-09-20:
> @muyir123 会的，修复已经在做内部数据集验证。因为修复影响范围较大，所以内部验证时间较久

好的，谢谢，麻烦尽快修复哈，这个BUG太久了

QingZhangBambu, 2024-10-16:
@muyir123 您好，这个问题已经修复，在下一个1.10 beta1可以拿到。请稍等。

muyir123, 2024-10-17:
> @muyir123 您好，这个问题已经修复，在下一个1.10 beta1可以拿到。请稍等。

好的，谢谢，我下载试试

muyir123, 2024-10-21:
> @muyir123 您好，这个问题已经修复，在下一个1.10 beta1可以拿到。请稍等。

咨询下您正式版合入这个功能大概还有多久呢？测试板不能发布makerworld上，电脑上同时存在正式版和测试板配置乱了

QingZhangBambu, 2024-10-22:
@muyir123 大约半个月到一个月内会陆续发布后续beta以及1.10正式版本