bambulab/BambuStudio #4258: Wrong Bridge Direction, printing a bridge on air

Orel-A, 2024-06-11:
### Bambu Studio Version

1.9.2 Public Official

### Where is the application from?

Bambu Lab github releases

### OS version

Windows 11

### Printer

Creality Ender-3 V3 KE

### How to reproduce

Bridge direction in settings is left on 0 as default.
A project file is provided.

### Actual results

The project file has 4 bridges, 2 are correct and 2 are in wrong direction, printing on air.

[image]

### Expected results

[image]

The expected results is to bridge from solid print.

### Project file & Debug log uploads

[GPU support jack (2).zip](https://github.com/user-attachments/files/15791122/GPU.support.jack.2.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-06-13:
@XunZhangBambu check this please

XunZhangBambu, 2024-06-17:
@Orel-A  I have found the root cause of the bug, will fix it later

NicolasBadey, 2024-06-19:
I confirm this behavior on new version, half or our products is no more printable due to this, do you have a release date @XunZhangBambu for the fix ?

It's like bridge event at "0" direction aren't smart now and follow always the same direction and yes, print in void (no more to the optimal automatic direction deduce from geometric).

ScrapyCocos, 2024-08-07:
I have the same issue. I'm currently printing a 90 degree part and one side where the slicer needed to print a bridge, the bridge it's printed in mid air and along the length of the design instead a proper bridge between two walls.

NicolasBadey, 2024-08-18:
It's like on previous version "direction = 0" was not typed (0 = false), and now it's typed so 0 is now "0 degree" and there is no alternative way to said "direction = auto".

This problem is critical as some pieces are not able to be print with BambuLab Slicer since

magnusmyklebost, 2024-10-28:
Any fix on this? I got two parts where the bridge is on the same height. But it still makes the right side bridge very long. I want to be able to control this better, and also that the software automatically detects how bridge length can be minimized.
[image]

Orel-A, 2024-10-28:
It's not actually 0 degrees, because in this case 0 means Automatic.

magnusmyklebost, 2024-10-28:
yea, that whas what i ment by "overwritten" even though 180 degrees should be the same direction as 0 (there is also a hint about this when you hover over the setting).

XunZhangBambu, 2024-10-29:
@magnusmyklebost  Hi,could you provide the 3MF. We have been working internally to fix the issue. In @Orel-A 3mf, the fixed auto bridge direction is 
[image]
However, there is a small gap at its corner.We are still optimizing this solution.

XunZhangBambu, 2024-11-18:
@magnusmyklebost  @ScrapyCocos  Could please provide the 3mf ? I need these 3MF files to verify if the fix is effective.

NicolasBadey, 2024-11-19:
@XunZhangBambu is it possible to have this beta to check if it works on my case ? Thanks
I can provide 3MF if you want

XunZhangBambu, 2024-11-20:
@NicolasBadey Sorry, I may not be able to provide the beta version for you to test. However, I can send you screenshots of the new results. I look forward to receiving your 3mf for testing!

ScrapyCocos, 2024-11-20:
> @magnusmyklebost @ScrapyCocos Could please provide the 3mf ? I need these 3MF files to verify if the fix is effective.

[LEDAirVent(2).zip](https://github.com/user-attachments/files/17837783/LEDAirVent.2.zip)

NicolasBadey, 2024-11-20:
@XunZhangBambu Yes, I isolate it thanks !
[test.zip](https://github.com/user-attachments/files/17837930/test.zip)
[image]