bambulab/BambuStudio #4261: Cannot build on Ubuntu 24.04 LTS

logiclrd, 2024-06-11:
### Bambu Studio Version

Git head

### Where is the application from?

Built from source by yourself or others

### OS version

Ubuntu 24.04 LTS

### Printer

n/a

### How to reproduce

Following the steps in "Linux Compile Guide", I cannot complete a build because it can't find OpenVDB.

### Actual results

```
CMake Deprecation Warning at cmake/modules/FindOpenVDB.cmake:126 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.
Call Stack (most recent call first):
  CMakeLists.txt:552 (find_package)

-- Could NOT find OpenVDB (missing: OpenVDB_LIB_COMPONENTS openvdb) (found suitable version "10.0.1", minimum required is "5.0")
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:317 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:552 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:324 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "10" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:552 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:345 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:367 (just_fail)
  CMakeLists.txt:552 (find_package)

CMake Error at CMakeLists.txt:557 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.

-- Configuring incomplete, errors occurred!
```

### Expected results

Successful build.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-06-12:
@logiclrd are you using the newest codes 1.9.2.57?

logiclrd, 2024-06-12:
As indicated in the report, I am using Git head. Commit 2a7368776b05517691a17947834af325cab42d97 to be exact.

logiclrd, 2024-06-12:
I will try checking out the v01.09.02.57 tag, see if that makes a difference.

logiclrd, 2024-06-12:
I get the same error with the repository on commit ba19c8dd68e2570b0a1097f1ad604ae5b62193f6, which is tag v01.09.02.57.

MackBambu, 2024-06-12:
> I get the same error with the repository on commit [ba19c8d](https://github.com/bambulab/BambuStudio/commit/ba19c8dd68e2570b0a1097f1ad604ae5b62193f6), which is tag v01.09.02.57.

Can you test it with my branch, I've recently made changes to cmake but haven't finished it yet.

logiclrd, 2024-06-12:
I'll give it a go, thanks :-)

logiclrd, 2024-06-12:
@MackBambu I merged your branch onto HEAD and I still get the same build error. :-(

MackBambu, 2024-06-13:
I added ubuntu 24.04 github actions.
https://github.com/MackBambu/BambuStudio/actions/runs/9501157711/job/26186013747

logiclrd, 2024-06-13:
The "Linux Compile Guide" says to run this command:
```
sudo apt-get install cmake clang git g++ build-essential libgl1-mesa-dev m4 libwayland-dev libxkbcommon-dev wayland-protocols extra-cmake-modules pkgconf libglu1-mesa-dev libcairo2-dev libgtk-3-dev libsoup2.4-dev libwebkit2gtk-4.0-dev libgstreamer1.0-dev libgstreamer-plugins-good1.0-dev libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad libosmesa6-dev
```
This doesn't actually work on Ubuntu 24.04. I switched out `libwebkit2gtk-4.0-dev` to `libwebkit2gtk-4.1-dev`. Could this be related??

MackBambu, 2024-06-14:
> The "Linux Compile Guide" says to run this command:
>
> ```
> sudo apt-get install cmake clang git g++ build-essential libgl1-mesa-dev m4 libwayland-dev libxkbcommon-dev wayland-protocols [...]
In order to install libwebkit2gtk-4.0-dev, it is necessary to add the source.This is the current temporary solution.
https://github.com/MackBambu/BambuStudio/blob/master/BuildLinux.sh#L111

logiclrd, 2024-06-14:
Is it a good idea to add a `jammy` source to a `noble` installation? Is it not possible to just build against the new minor version number??

MackBambu, 2024-06-14:
> Is it a good idea to add a `jammy` source to a `noble` installation? Is it not possible to just build against the new minor version number??
Using the new minor version number will require changes to the wxWidget, which are not yet ready

logiclrd, 2024-06-14:
Okay, well, I went to the Ubuntu Packages web site and checked the dependency tree, it was only 6 packages. I downloaded them and extracted the archives manually, and all of the files are either versioned or in versioned folders. So, I installed the 6 packages. Trying to build again now.

logiclrd, 2024-06-14:
Unfortunately, the build still fails with the same error:
```
-- Could NOT find OpenVDB (missing: OpenVDB_LIB_COMPONENTS openvdb) (found suitable version "10.0.1", minimum required is "5.0")
CMake Warning at cmake/modules/OpenVDBUtils.cmake:129 (message):
  vdb_print not found! OPENVDB_PRINT-NOTFOUND
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:318 (OPENVDB_ABI_VERSION_FROM_PRINT)
  CMakeLists.txt:557 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:325 (message):
  Unable to determine OpenVDB ABI version from OpenVDB installation.  The
  library major version "10" will be inferred.  If this is not correct, use
  add_definitions(-DOPENVDB_ABI_VERSION_NUMBER=N)
Call Stack (most recent call first):
  CMakeLists.txt:557 (find_package)

CMake Warning at cmake/modules/FindOpenVDB.cmake:346 (message):
  IlmBase::Half can not be found!
Call Stack (most recent call first):
  cmake/modules/FindOpenVDB.cmake:368 (just_fail)
  CMakeLists.txt:557 (find_package)

CMake Error at CMakeLists.txt:562 (message):
  OpenVDB could not be found with the bundled find module.  You can try to
  specify the find module location of your OpenVDB installation with the
  OPENVDB_FIND_MODULE_PATH cache variable.
```

logiclrd, 2024-06-14:
I went and found the OpenVDB code, cloned the repository and built it from code, and I've been able to get past that error. Now I'm stuck on:
```
-- IGL NOT found, using bundled version...
-- [MCUT] version: 1.2.0
-- [MCUT] MCUT_INCLUDE_DIR=/code/BambuStudio/src/mcut/include
-- [MCUT] compilation_flags=-Wall;-Wextra
-- [MCUT] preprocessor_defs=-DMCUT_WITH_COMPUTE_HELPER_THREADPOOL=1
-- [MCUT] extra_libs=Threads::Threads
-- [MCUT] create target: name=mcut type=STATIC
-- [MCUT] MCUT_LIB_PATH=$<TARGET_FILE:mcut>
-- Using header-only CGAL
-- Targetting Unix Makefiles
-- Using /usr/bin/c++ compiler.
CMake Error at /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find MPFR (missing: MPFR_LIBRARIES MPFR_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  deps/build/dep_CGAL-prefix/src/dep_CGAL/Installation/cmake/modules/FindMPFR.cmake:52 (find_package_handle_standard_args)
  deps/build/dep_CGAL-prefix/src/dep_CGAL/Installation/cmake/modules/CGAL_SetupGMP.cmake:25 (find_package)
  deps/build/dep_CGAL-prefix/src/dep_CGAL/Installation/cmake/modules/CGAL_SetupCGALDependencies.cmake:36 (include)
  deps/build/dep_CGAL-prefix/src/dep_CGAL/Installation/lib/cmake/CGAL/CGALConfig.cmake:153 (include)
  deps/build/dep_CGAL-prefix/src/dep_CGAL/CGALConfig.cmake:6 (include)
  src/libslic3r/CMakeLists.txt:435 (find_package)

-- Configuring incomplete, errors occurred!
```
I haven't been able to figure out how to point it at the MPFR library that the "building the deps" step put into the `deps` folder.

MackBambu, 2024-06-14:
I suggest you try to compile with BuildLinux.sh.It might be easier.
```
sudo ./BuilLinux.sh -ur //instatll package
sudo ./BuildLinux.sh -dr //compile dependencies
sudo ./BuildLinux.sh -ir //compile studio
```

logiclrd, 2024-06-14:
Hmm, `BuildLinux.sh -i` doesn't seem to build Bambu Studio. The help text says that it should "Generate appimage", but after doing step [2/9] it just exits.

If I `BuildLinux.sh -s` then it proceeds to step [7/9] and then encounters the MPFR library error.

MackBambu, 2024-06-14:
When you build the dependency ‘-dr’, whether you encountered some errors, when building the dependency encountered errors will not affect the continuation of the build, you need to check the log.

logiclrd, 2024-06-14:
Here's the full log:
```
root@visor:/code/BambuStudio# ./BuildLinux.sh -dr
FOUND_GTK2=rc  libgtk2.0-0:amd64          2.24.33-2ubuntu2   amd64        GTK graphical user interface library - old version
ii  libgtk2.0-0t64:amd64       2.24.33-4ubuntu1   amd64        GTK graphical user interface library - old version
ii  libgtk2.0-bin              2.24.33-4ubuntu1   amd64        programs for the GTK graphical user interface library
ii  libgtk2.0-common           2.24.33-4ubuntu1   all          common files for the GTK graphical user interface library)
[1/9] Updating submodules...
/code/BambuStudio/resources/profiles /code/BambuStudio
/code/BambuStudio
[2/9] Changing date in version...
done
[3/9] Configuring dependencies...
/code/BambuStudio/deps/build /code/BambuStudio
CMake Warning (dev) at CMakeLists.txt:23 (project):
  cmake_minimum_required() should be called prior to this top-level project()
  call.  Please see the cmake-commands(7) manual for usage documentation of
  both commands.
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Deprecation Warning at CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- BambuStudio deps DESTDIR: /code/BambuStudio_dep
-- BambuStudio dowload dir for source packages: /code/BambuStudio/deps/build
-- BambuStudio deps debug build: ON
CMake Warning (dev) at /usr/share/cmake-3.28/Modules/ExternalProject.cmake:3195 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/ExternalProject.cmake:4418 (_ep_add_download_command)
  CMakeLists.txt:80 (ExternalProject_Add)
  PNG/PNG.cmake:13 (bambustudio_add_cmake_project)
  CMakeLists.txt:152 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Deduced boost toolset: gcc based on GNU compiler
CMake Warning (dev) at /usr/share/cmake-3.28/Modules/ExternalProject.cmake:3195 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
  preserves the timestamps from the archive instead, but this is usually not
  what you want.  Update your project to the NEW behavior or specify the
  DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
  robustness issue.
Call Stack (most recent call first):
  /usr/share/cmake-3.28/Modules/ExternalProject.cmake:4418 (_ep_add_download_command)
  Boost/Boost.cmake:129 (ExternalProject_Add)
  CMakeLists.txt:162 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/share/cmake-3.28/Modules/ExternalProject.cmake:3195 (message):
  The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
  not set.  The policy's OLD behavior will be used.  When using a URL
  download, the timestamps of extracted files should preferably be that of
  the time of extraction, otherwise code that depends on the extracted
  contents might not be rebuilt if the URL changes.  The OLD behavior
[... 2020 lines cut ...]
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_core-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_adv-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_aui-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_html-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_propgrid-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_ribbon-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_richtext-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_webview-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/wx/3.1.5/web-extensions/webkit2_extu-3.1.so
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_xrc-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_media-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_gl-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_gtk3u_qa-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/lib/libwx_baseu_xml-3.1.a
-- Up-to-date: /code/BambuStudio_dep/usr/local/bin/wxrc
[ 97%] Completed 'dep_wxWidgets'
[100%] Built target dep_wxWidgets
[100%] Built target deps
done
/code/BambuStudio
done
root@visor:/code/BambuStudio# 
```

I don't see any errors, but I copied the whole log because you probably have sharper eyes than I do. :-)

MackBambu, 2024-06-15:
Did you modify the BuildLinux.sh file? The default installation directory looks completely different. My guess is that the dependency directory was incorrectly.

logiclrd, 2024-06-16:
There was a small change that I made while trying to sort out OpenVDB issues:
```
root@visor:/code/BambuStudio# git diff BuildLinux.sh 
diff --git a/BuildLinux.sh b/BuildLinux.sh
index 5be61155..3aefed07 100755
--- a/BuildLinux.sh
+++ b/BuildLinux.sh
@@ -226,6 +226,8 @@ then
     else
         BUILD_ARGS="${BUILD_ARGS} -DBBL_RELEASE_TO_PUBLIC=1 -DBBL_INTERNAL_TESTING=0"
     fi
+
+    BUILD_ARGS="${BUILD_ARGS} -DOPENVDB_FIND_MODULE_PATH=/usr/lib/x86_64-linux-gnu/cmake/OpenVDB"
     
     # cmake
     pushd build
root@visor:/code/BambuStudio#
```
I reverted this change but I still get the same result. I'm not sure what you mean by installation directory -- can you be more specific? If you're referring to the dependencies being in `/code/BambuStudio_dep` instead of my home directory, that's because earlier, when following the Linux Compile Guide, I specified that path instead of a path inside my home directory to `cmake`, I'm pretty sure:
```
# From the Linux Compile Guide:
cmake ../ -DDESTDIR="/home/_username_/work/projects/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1

# My command:
cmake ../ -DDESTDIR="/code/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1
```

logiclrd, 2024-06-16:
I just cleared the `build` folder and started from scratch, going straight to `BuildLinux.sh`. Now it fails trying to find NLopt.

MackBambu, 2024-06-16:
If you build with BuildLinux.sh, his dependency directory is in deps/build/destdir/usr/local.
I suggest you refer to the github actions workflow build process.
https://github.com/MackBambu/BambuStudio/tree/master/.github/workflows

logiclrd, 2024-06-16:
Hmm, that workflow definition runs on Ubuntu 20.04, which is 4 major versions behind what I'm running. This squares with my experience of immediately running into dependencies issues; not all of the same packages are present in 24.04 noble.

MackBambu, 2024-06-16:
sorry，i sent the wrong link.
https://github.com/MackBambu/BambuStudio/tree/master/.github/workflows
https://github.com/MackBambu/BambuStudio/actions/runs/9530394294

logiclrd, 2024-06-16:
Hey, so I can just use that build artifact and it'll have a usable binary? :-P Still don't know why my build experience is different, but in this instance I'm not terribly worried if I have something I can run.

MackBambu, 2024-06-16:
Yes, they're exactly the same, everyone's system comes pre-installed in a different environment and will always have different issues

logiclrd, 2024-06-17:
Cool cool, it works. Guess I'll close this issue. The issue isn't resolved but the underlying tension is. Thanks :-)

closed 2024-06-17 by logiclrd