bambulab/BambuStudio #4262: Unable to login to both Bambu Studio and Bambu Handy at the same time

Benjii43, 2024-06-11:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

GPU: NVIDIA GeForce GTX 1050 Ti
CPU: AMD FX(tm)-6300 Six-Core Processor

### Printer

Bambu Lab P1P

### How to reproduce

[image]

### Actual results

I am trying to login to both the bambu hand and bambu studio at the same time. Bambu handy works just fine, but the bambu studio wont let me log in or find the device manually using the "bind" system. I put in the code on the printer and nothing happens. I am just at this blank screen on the home page. 

### Expected results

I am expecting to be able to login to both bambu handy and bambu studio. 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Benjii43, 2024-06-11:
[debug_Tue_Jun_11_18_25_42_12140.log.zip](https://github.com/user-attachments/files/15795688/debug_Tue_Jun_11_18_25_42_12140.log.zip)

bambu123, 2024-06-12:
Please refer to the wiki and give us feedback
https://wiki.bambulab.com/en/bambu-studio/troubleshoot/install-issue