bambulab/BambuStudio #4264: Changed code signature in macOS version 1.9.2?

suschizu, 2024-06-11:
### Bambu Studio Version

1.9.2

### Where is the application from?

Bambu Lab github releases

### OS version

macOS 13.6

### Printer

Bambu Lab

### How to reproduce

codesign -dv --verbose=4 /Path/To/Application.app

### Actual results

FOUND 'Shanghai Lunkuo Technology Co., Ltd (T3UBR9Y3B2)'

### Expected results

EXPECTED 'Shenzhen Tuozhu Technology Co., Ltd (F4SKHPXDD9)' but 

### Project file & Debug log uploads

[image]

### Checklist of files to include

- [X] Log file
- [ ] Project file

lanewei120, 2024-06-12:
@suschizu yes, we switched to the new signature "Shanghai Lunko..."
which is a subsidiary of “Shenzhen TuoZhu..."

suschizu, 2024-06-12:
thanks for the fast clarification!

closed 2024-06-12 by suschizu

JeremyLaurenson, 2024-06-13:
This should be reopened. My Mac will no longer run Bambu Studio properly as a result of this. There needs to be a help article on how to fix because most Mac users will do the right thing and get locked out of the app as a result of it being flagged as changed.

lanewei120, 2024-06-14:
> This should be reopened. This change will cause an urgent issue for just _about all Mac users on a recent MacOS load_. If you do the default thing which is not to trust the change (Which most users should do) the app will no longer run properly on the Mac. My Mac will no longer run Bambu Studio properly as a result of this.
> 
> There needs to be a help article on how to fix because most Mac users will do the right thing and get locked out of the app as a result of it being flagged as changed.

could you give a video to show us the detailed issue?
we have not met any issues when switch to this signature