bambulab/BambuStudio #4266: Faulty Power Switch

tapowe-beep, 2024-06-12:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Printer

A1

### How to reproduce

I ordered the A1 last December...I did the heatbed replacement myself and everything worked fine. I've just noticed over the past few weeks that sometimes the printer switches itself off. When I check it, the switch is on, but the printer is off. If I touch or wiggle the switch, the printer turns on. I took off the bottom of the printer and checked the cabling and wires around the power switch and everything seems secure, no loose wires. I have another A1 that doesn't have this issue.

### Actual results

I think the above is self-explanatory

### Expected results

I think the above is self-explanatory

### Project file & Debug log uploads

This isn't project file dependent, it's a hardware issue.

### Checklist of files to include

- [X] Log file
- [X] Project file

muyir123, 2024-06-12:
这个不应该在这里反馈，在APP里提交工单或者联系购买平台客服就行
This should not be reported here. Just submit a work order in the app or contact customer service on the purchasing platform

EhRabz, 2024-06-12:
This isn't the support site dude..... Delete this and go contact support lol

closed 2024-06-12 by tapowe-beep