bambulab/BambuStudio #4269: bambu studio 切片时总是崩溃闪退

ClamDinner, 2024-06-12:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website, Bambu Lab github releases

### OS version

Windows11

### Additional system information

CPU:13代酷睿I5 13600K
显卡:Radeon RX 590 Series
内存:16GB

### Printer

Bambu Lab A1

### How to reproduce

1、点击切片单盘或预览

### Actual results

程序崩溃闪退并直接关闭

### Expected results

正常切片完成

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15803844/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

QingZhangBambu, 2024-06-14:
https://github.com/bambulab/BambuStudio/issues/3318
你好，超频的CPU对多线程的支持有时候不大稳定，所以会造成闪退。可以尝试一下这里的方法来保证允许稳定。

ClamDinner, 2024-06-14:
> #3318 你好，超频的CPU对多线程的支持有时候不大稳定，所以会造成闪退。可以尝试一下这里的方法来保证稳定。

CPU频率调整成自动之后可以了，感谢

closed 2024-06-17 by QingZhangBambu