bambulab/BambuStudio #4273: Bambu crashing on Slice

Duecki1, 2024-06-12:
### Bambu Studio Version

1.9.1.66

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 11

### Additional system information

Doesn't matter, happend on my PC and my Surface Laptop running both win11

### Printer

P1P

### How to reproduce

I sliced my file and go errors. Im using a 0.2mm nozzle, organic tree supports, line width of 0.1, print by object and the print is going to take 1d.

### Actual results

Crashes not only the programm, but my whole system and i need to restart
[image]
[image]

### Expected results

Slice the file

### Project file & Debug log uploads

no

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-06-14:
Hi, could you share your 3mf file?

QingZhangBambu, 2024-06-14:
https://github.com/bambulab/BambuStudio/issues/3318 and These methods may help

Duecki1, 2024-06-14:
> Hi, could you share your 3mf file?

No sorry, due to the model being a paid model.

Duecki1, 2024-06-14:
> https://github.com/bambulab/BambuStudio/issues/3318 and These methods may help

Thanks, but I just downgraded to version 1.7 from January. It also seems to be better with support xD. Thanks tho

tangketan, 2024-06-17:
It might be due to organic support. From 1.9 we have changed the default support style to organic tree. Please try manually setting the tree style to "tree hybrid" and see if slicing is OK.
[image]

tangketan, 2024-06-17:
By the way, the log files might help. Please upload them if you have any, which are safe. Or you can email me the log files privately.

closed 2026-08-30 by Duecki1