bambulab/BambuStudio #4274: Bambu Slicer Crashes when filing bed with clones

Nschus0, 2024-06-12:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

windows 11

### Additional system information

13th gen intel i7-13700KF
DRAM 32GB @3200mHz
NVIDIA GeForce RTX 3080

### Printer

Bambu X1C

### How to reproduce

Load in file, select fill bed with clones

### Actual results

Bambu slicer crashes with no report. Can duplicate them manually but then when going to slice also crashes

### Expected results

Ready to print objects

### Project file & Debug log uploads

[crash_Wed_Jun_12_17_15_13_0_done.log](https://github.com/user-attachments/files/15811547/crash_Wed_Jun_12_17_15_13_0_done.log)
[Socket Holder.zip](https://github.com/user-attachments/files/15811587/Socket.Holder.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Nschus0, 2024-06-12:
Can slice individual models no issue but cannot slice more then a small handful of objects at a time.

QingZhangBambu, 2024-06-14:
@LiZ-Li-BBL check this please

LiZ-Li-BBL, 2024-06-17:
Could you provide the crash dmp file in the BambuStudio/log directory?
[image]

QingZhangBambu, 2024-06-17:
> Can slice individual models no issue but cannot slice more then a small handful of objects at a time.

Could open another issue for slicing crash? and upload 3mfs and dump files?

rrowhe4d, 2024-06-24:
I'm having the exact same problem, version 1.9.3.50, a single object works fine, but when filling the bed after cloning, I get a program crash when clicking the preview button. Please fix.

QingZhangBambu, 2024-06-24:
@Nschus0 
https://github.com/bambulab/BambuStudio/issues/3318
Hi, could you try these methods? sometimes the CPU may not be stable enough to support multithreading.