bambulab/BambuStudio #4275: Unable to select PA profile or enter Factor K

patrickcheah, 2024-06-13:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Additional system information

Apple M1 Pro MacBook Pro 2021

### Printer

Bambu P1S

### How to reproduce

In the Device tab, interacting on the filaments on my AMS:
1) Factor K has no value
2) Drop down for PA profile is not working
[image]
[image]

### Actual results

In the Device tab, interacting on the filaments on my AMS:
1) Factor K has no value
2) Drop down for PA profile is not working
[image]
[image]

### Expected results

1) Factor K value should be present and I can update it
2) PA profile should be available to be selected with the dropdown

### Project file & Debug log uploads

[Uploading Magsafe Phone and Watch Stand(2).3mf.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

gitbreiti, 2024-06-14:
Same problem here. Seems to have come with the new firmware update (01.06.00.00) for P1P/P1S

TaylorN15, 2024-06-14:
> I have found it. It is not a firmware error. You can create a profile under Dynamic Flow Calibration and then select it.

I found this also, but my issue is that you have to create a profile for every type of filament you have. Even though I have done this before, it wasn't saved like now. I would prefer to be able to choose a profile _or_ enter a value manually.

TinyMito, 2024-06-17:
I just updated Bambu Studio 1.9.2.57 and latest beta firmware on P1S... it lost all the preset K values under Manage Result in Calibration tab page.

zhimin-zeng-bambulab, 2024-06-28:
The new P1P/S firmware is the same as the X series. It stores the k value on the machine and manages it through the "Manage Results" of Calibration page. If you already know the k value, you can add it directly here.
[image]

fatboy1271, 2024-08-08:
Thank you @zhimin-zeng-bambulab! After going to Calibration > Manage Result I was able to select it 🤘🤘

tjc2005, 2024-08-28:
This is still happening, very frustrating!

zhimin-zeng-bambulab, 2024-08-29:
@tjc2005  What is your problem? Could you describe it in detail? Can't the above method solve your problem?

tjc2005, 2024-08-29:
> @tjc2005 What is your problem? Could you describe it in detail? Can't the above method solve your problem?

Drop down for PA profile is not working in AMS menu.

I have to close Bambu studio and reopen, sometimes that works, but it seems to be luck of the draw wether the drop down menu decides to work or not.

JonasKuschel, 2024-10-08:
I have the same bug. I have setup the k-values in the Calibration Tab, however whenever I restart the computer or printer or idk when this happens. The results just won't show up. When I go to Calibration > Manage Results and wait for a short time until the Calibrations are loaded, I can go back to the device tab and now they become visible.

OzDruiD, 2025-11-14:
This is also now happening to me .. anyone found a better fix that going the  workaround

Workaround-
"Calibration > Manage Results and wait for a short time until the Calibrations are loaded, I can go back to the device tab and now they become visible.

GKMoritz, 2026-01-16:
@hemai249 This is definitely not fixed as I had this issue again just today all the PA profiles are gone after restarting PC and Printer. Very frustrating and a waste of filament.