bambulab/BambuStudio #4276: Crate custom filament - not working when based on "copy of current filament"

Goger-TKL, 2024-06-13:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Printer

P1S with AMS

### How to reproduce

Add a custom filament, select based on copy of current filament. Click OK. Error message:
[image]

It is not possible to select a printer in this case.

### Actual results

Not possible to craete the filametn because no printer can be selected

### Expected results

Be able to select a printer and create custom filament

### Project file & Debug log uploads

No project needed.
[log.zip](https://github.com/user-attachments/files/15817486/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-06-19:
Hello, I have reviewed the log you sent and found that there may be an issue with the preset "Strongarm PETG Basic @ Bambu Lab P1S 0.4 noise". Could you please send it to me. You can search in the same level folder as the log, and its file name is "Strongarm PETG Basic @ Bambu Lab P1S 0.4 noise". (In the Filament folder of users in the same level of log folder)

Goger-TKL, 2024-06-20:
Hi. Thank you for your help. There is no file with this name. I suppose you mean the file with nozzle at the end. I will attach it.

Also one question: Is it possible to rename a filament brand as I made a mistake there or do I have to make a new one?

[Strongarm PETG Basic @Bambu Lab P1S 0.4 nozzle.json](https://github.com/user-attachments/files/15909558/Strongarm.PETG.Basic.%40Bambu.Lab.P1S.0.4.nozzle.json)

DanBao-Bambu, 2024-06-26:
[image]
Thank you very much for sending the file. I found that one of the "compatible_printers" fields is empty. I don't know why this field is empty, but I know the reason why you can't choose a printer at this time. Do you still remember which material your custom material was created based on? I see that the preset version is 1.8.0.25, and I think you can create a custom Filament again. Note that the machine types that can be selected when creating custom Filaments are also related to the printer you choose on the printer wizard page, and only the printer you choose will be displayed.

Goger-TKL, 2024-06-27:
It should be based on generic petg.