bambulab/BambuStudio #4277: Model auto orientation & alignment drop models out of table! but space enough to place

iZonex, 2024-06-13:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab github releases

### OS version

MacoS Sonoma 14.5

### Additional system information

M2 Pro, 16 Gb

### Printer

X1C

### How to reproduce

Added different models to table what can fill all table
Click to Auto Align and Auto Rotate

### Actual results

Models out of the build plate

### Expected results

Models on the build plate

### Project file & Debug log uploads

NDA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tangketan, 2024-06-14:
3mf file please.

darkneon2002, 2024-06-18:
Yeah it's terrible at doing these things

iZonex, 2024-07-30:
@tangketan 
[demo.3mf.zip](https://github.com/user-attachments/files/16423434/demo.3mf.zip)