bambulab/BambuStudio #4278: BambuLab Software Codesign has changed. This will break most Macs

JeremyLaurenson, 2024-06-13:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

Bambu X1P 

### How to reproduce

Downloading and executing the latest build from the official website causes MacOS security to block it by default because of a code signing developer ID change:

Execute the command: codesign -dv --verbose=4 /Applications/BambuStudio.app
Against previous versions and the current version
This change is developerID leads tot he app being blocked on MacOS by default. Once its blocked I do not know how to unblock it.

### Actual results

Identifier=com.bambulab.bambu-studio
Format=app bundle with Mach-O universal (x86_64 arm64)
CodeDirectory v=20500 size=814949 flags=0x10000(runtime) hashes=25456+7 location=embedded
VersionPlatform=1
VersionMin=659200
VersionSDK=852224
Hash type=sha256 size=32
CandidateCDHash sha256=fdb47f9a28c88a0a775effec2009cbd9624b4a14
CandidateCDHashFull sha256=fdb47f9a28c88a0a775effec2009cbd9624b4a142bd5409af602c72460706835
Hash choices=sha256
CMSDigest=fdb47f9a28c88a0a775effec2009cbd9624b4a142bd5409af602c72460706835
CMSDigestType=2
Executable Segment base=0
Executable Segment limit=71860224
Executable Segment flags=0x1
Page size=4096
CDHash=fdb47f9a28c88a0a775effec2009cbd9624b4a14
Signature size=9020
Authority=Developer ID Application: Shanghai Lunkuo Technology Co., Ltd (T3UBR9Y3B2)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Jun 7, 2024 at 8:17:50 AM
Info.plist entries=16
TeamIdentifier=T3UBR9Y3B2
Runtime Version=13.1.0
Sealed Resources version=2 rules=13 files=2921
Internal requirements count=1 size=188

### Expected results

Identifier=com.bambulab.bambu-studio
Format=app bundle with Mach-O universal (x86_64 arm64)
CodeDirectory v=20500 size=772133 flags=0x10000(runtime) hashes=24118+7 location=embedded
VersionPlatform=1
VersionMin=659200
VersionSDK=852224
Hash type=sha256 size=32
CandidateCDHash sha256=3ba61a9fb02772be477856eaa0f8155dc311136b
CandidateCDHashFull sha256=3ba61a9fb02772be477856eaa0f8155dc311136b8a15ac67fca29c15172e0dd7
Hash choices=sha256
CMSDigest=3ba61a9fb02772be477856eaa0f8155dc311136b8a15ac67fca29c15172e0dd7
CMSDigestType=2
Executable Segment base=0
Executable Segment limit=67534848
Executable Segment flags=0x1
Page size=4096
CDHash=3ba61a9fb02772be477856eaa0f8155dc311136b
Signature size=9021
Authority=Developer ID Application: Shenzhen Tuozhu Technology Co., Ltd (F4SKHPXDD9)
Authority=Developer ID Certification Authority
Authority=Apple Root CA
Timestamp=Dec 4, 2023 at 5:36:16 AM
Info.plist entries=16
TeamIdentifier=F4SKHPXDD9
Runtime Version=13.1.0
Sealed Resources version=2 rules=13 files=2559
Internal requirements count=1 size=188

### Project file & Debug log uploads

No project files.

### Checklist of files to include

- [X] Log file
- [X] Project file

JeremyLaurenson, 2024-06-13:
When users see this dialog:

[image]

they must click "Accept Modification" or BambuStudio will be denied access to any network.

If the user navigates away or closes the app they will not see this message again in order to be able to accept the modification, unless they uninstall the app (drag to trash, empty trash) AND reboot the Mac and reinstall the Bambu Studio app.

lanewei120, 2024-06-14:
@JeremyLaurenson 
hello, we have tried on lots of Macos laptops, can not meet this issue
Do you know what this issue is related to?
maybe os version?

swheettaos, 2024-06-14:
I can't get the latest version to run also on my MacOS.  I never saw the message in the above screenshot.  I'm on a MBP with intel (last model that had Intel chip) and the app will launch but no main screen is presented.  I tried going back to older version it has the same problem.  I was running the latest version of Sonoma when I had this problem to start with and then updated to the latest beta of MacOS with no difference. 

I tired going back to older version, I tried completely removing it and re-installing it .. same problem I can't use the app.  On install it starts .. downloads the network plugin and then nothing.  No main screen .. no errors.. no nothing.

JeremyLaurenson, 2024-06-14:
> I can't get the latest version to run also on my MacOS. I never saw the message in the above screenshot. I'm on a MBP with intel (last model that had Intel chip) and the app will launch but no main [...]

Try removing entirely, rebooting, reinstalling.

swheettaos, 2024-06-14:
I deleted the app (deleted from application folder) and rebooted and then reinstalled.. Same issue.

swheettaos, 2024-06-14:
ok.. I was able to get it working .. it was some weird multi screen thing.  I was able to the menu and resize it and it appeared.. didn't work from the dock .. had to use the menu item.  Weird.

lanewei120, 2024-06-19:
> ok.. I was able to get it working .. it was some weird multi screen thing. I was able to the menu and resize it and it appeared.. didn't work from the dock .. had to use the menu item. Weird.

@swheettaos 
we have tried on lots of Macos laptops, can not meet this issue
to help us to debug
could you show us your OS and system information?

JeremyLaurenson, 2024-06-19:
The solution for this was to remove the app, reboot and re-install the app

Then when you get prompted with the issue:

￼

You MUST click “Accept modification” or it blocks Bambu Studio network access

lanewei120, 2024-06-19:
> The solution for this was to remove the app, reboot and re-install the app Then when you get prompted with the issue: ￼ You MUST click “Accept modification” or it blocks Bambu Studio network access
> […](#)

Hello @JeremyLaurenson 
do you know any clue on how this issue can be reproduced?
we have updated our macos to 14.5 today
and still can not reproduce this issue

JeremyLaurenson, 2024-06-19:
> … <x-msg://4/#>

I would use an Intel Mac then:
Remove Bambu Studio, reboot.
Install an older version with the older code sign, and enable network.
Then download current version, copy over old and launch it

Thats what gives me this error

Aaron-Rumpler, 2024-06-22:
@JeremyLaurenson The dialog you're seeing is from [Little Snitch](https://www.obdev.at/products/littlesnitch/index.html) (a 3rd-party firewall), not from macOS itself. This is only an issue for Little Snitch users.

JeremyLaurenson, 2024-06-22:
Aha! Thanks for the info, @Aaron-Rumpler 
Ill reach out to those guys to make sure they make this more obvious.

closed 2024-06-22 by JeremyLaurenson