bambulab/BambuStudio #4280: Every time I open somebody else's 3MF, color syncing with my AMS is wrong

icerabbitje, 2024-06-13:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Mac OS 14.4.1

### Printer

BambuLab X1C with AMS

### How to reproduce

Open a multi color file from MakerWorld, example any multi-color License Plate Tag

This loads with the authors filament, not the filament in my AMS. 
( 99% Chance I will never have the author's filaments ) 

Then syncing with my AMS results in the following screens, prompts and inability to actually print because roll 3 and 4 are incorrect. 

[image]

[image]

[image]

[image]

While this is what is in my machine, with right now slot 3 empty:

[image]

Why does Bambu Studio not match my machine??? 

Now I can manually switch roll 4 to blue, but I should not have to, should I.

( and I cannot declare 3 as empty in Bambu Studio ) 

### Actual results

Results as above ... 

### Expected results

SYNC my 4 AMS colors properly, every slot and color. 

### Project file & Debug log uploads

[licenseplate_v1(2) send to bambu.3mf.zip](https://github.com/user-attachments/files/15827621/licenseplate_v1.2.send.to.bambu.3mf.zip)
[debug_Fri_Jun_07_15_42_51_51094.log.0.zip](https://github.com/user-attachments/files/15827622/debug_Fri_Jun_07_15_42_51_51094.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Nicog77, 2024-06-14:
Same behaviour on P1S with AMS. Bambu Studio 1.9.2.57

DanBao-Bambu, 2024-06-19:
The synchronization result is expected, and the preset list of materials in Studio after synchronization is arranged in the order of AMS. Of course, when encountering an empty slot, Studio will use the material preset of the previous and next slots in AMS for that location, and finally add external materials. The Studio material list cannot display empty material presets. This issue will be resolved by mapping material presets to slot positions during printing.

icerabbitje, 2024-06-19:
This was like the third or fourth time running into this issue, and I am pretty sure that I had 4 slots filled before. Thought it was my error, but it is not. 

Requesting a SYNC on Bambu Studio should - in my humble opinion - copy exactly what it says on the AMS.  
Period. It should not error by mixing colors or saying it can't resolve something. 

And a user should not have to do it Bambu Studio. 

When I load my AMS, I tell my X1C what I loaded in.

Then I expect Bambu Studio to listen to what my X1C says is loaded in. 

Note that my X1C is not sitting next to the computer that I have Bambu Studio on.  I can't tell what is inside the AMS, without physically walking over to it. So I expect a SYNC to positively update what is in the AMS, no iffy stuff, no errors, ...  I shouldn't even matter if it is just one roll in slot 1 or one roll in slot 4, to give an extreme example.