bambulab/BambuStudio #4281: Cant slice with newest version a part

butch555, 2024-06-13:
### Bambu Studio Version

Bambu_Studio_win_public-v01.09.02.57-20240607194639

### Where is the application from?

Bambu Lab Official website

### OS version

win11

### Additional system information

14700k , 4080 , 32gb ddr5 6800

### Printer

4x P1S

### How to reproduce

Every time crash of studio while slicing @15 or 25%.....depends on parameter......with previous versions no prob......100 times printed the last month
[AEA Pumpaction Terminator Griff kurz 2.0.zip](https://github.com/user-attachments/files/15829405/AEA.Pumpaction.Terminator.Griff.kurz.2.0.zip)

### Actual results

crash to windows

### Expected results

slicing ok

### Project file & Debug log uploads

[Uploading AEA Pumpaction Terminator Griff kurz 2.0.zip…]()

### Checklist of files to include

- [ ] Log file
- [X] Project file

QingZhangBambu, 2024-06-14:
hI, I can't reproduce it. Could you share the dump file to help me target the issue?

butch555, 2024-06-14:
i will install the latest bambu studio a second time to check this....maybe the first install went wrong....can do it at sunday.....bambu only crashed when slicing in upright position. 20 times checked everything.

QingZhangBambu, 2024-06-14:
https://github.com/bambulab/BambuStudio/issues/3318
and if it still crashes while slicing,  you can try these methods. sometimes the CPU may not be stable enough to support multithreading.