bambulab/BambuStudio #4282: The "Scarf Joint Seam" when set to Contour shifts the PAUSE feature in Bambu Studio

zkcreations, 2024-06-14:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

macOS Sonoma 14.5

### Additional system information

MacBook Pro 16in 2021

### Printer

Bambu P1S

### How to reproduce

1) I enable the the "Scarf Joint seam (experimental)" and set it to Contour in Bambu studio with one of my 3D prints
2) I then click Slice plate.
3) I click to "add pause" to a specific line number

### Actual results

I check the g-code to see if the pause is placed in the correct location but it is shifted to 2 layers above
[image]

### Expected results

The Pause should be 2 layers before this.

### Project file & Debug log uploads

Sorry I don't think you need this you can easily see the problem from the attached image. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

QingZhangBambu, 2024-06-17:
it had been fixed. Could you try the latest version?

zkcreations, 2024-06-17:
Hello,I updated the software today and yes it fixed the problem. Thank you,ZviSent from my iPad

closed 2024-06-17 by QingZhangBambu