bambulab/BambuStudio #4283: Bed exclude are limits incorrect

HuberDe, 2024-06-14:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

Anycubic Chiron

### How to reproduce

1. Add a Anycubic Chiron
2. Define a bed size of 410x430
3. Add the bed exclude "0x0, 410x0, 410x40, 0x40, 0x380, 410x380, 410x430, 0x430"

### Actual results

Error message, that the exclude area is out of range. The maximum size of the exclude area is: "0x0, 200x0, 200x40, 0x40, 0x150, 200x150, 200x200, 0x200" which results in the following picture:

[image]

Btw: The rendered area of the active plate is incorrect drawn:

[image]

### Expected results

A defined bed exclude area excluding the first and last 40mm (y) over the complete x axis

### Project file & Debug log uploads

No logs needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-06-17:
yes, currently exclude area doesn't support  multiple areas well

could you show us which printer are you using and  waht are the real exclude areas?

HuberDe, 2024-06-17:
Also one area doesn't support with a width higher than 200.

Im using a Chiron with a glass bed and bed clamps which should be excluded.