bambulab/BambuStudio #4288: Can't log in to Bambu studio on macOS via Sign in with Apple

spawnofbill, 2024-06-14:
### Bambu Studio Version

01.09.02.57

### Where is the application from?

Bambu Lab Official website

### OS version

macOS 15.0

### Additional system information

Apple M1 Max Mac Studio

### Printer

Bambu X1C

### How to reproduce

1. Try to sign in to Bambu Labs account with "Sign in with Apple"
2. Safari opens and successfully authentics the login with TouchID
3. Webpage loads a Bambu labs page, and then attempts to load a localhost page with the authentication token but safari returns "Safari Can't Find the Server" because "Safari can't find the server "localhost".

### Actual results

App fails to log into the the account created with "Sign in with Apple."

### Expected results

App receives token to sign into the account created with "Sign in with Apple" and signs into the associated account.

### Project file & Debug log uploads

[debug_Fri_Jun_14_19_25_46_59907.log.0.zip](https://github.com/user-attachments/files/15844815/debug_Fri_Jun_14_19_25_46_59907.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

rekonnected, 2024-07-13:
Also experiencing SIWA failure here, but it doesn't even get to the opening Safari part. It just crashes the instant I press the Apple icon. 

Studio v1.9.3.50, macOS Sonoma 14.5 (23F79), M1 Pro Macbook Pro. Downloaded from official site as well.
[image]
[debug_Sat_Jul_13_09_55_54_9536.log](https://github.com/user-attachments/files/16202379/debug_Sat_Jul_13_09_55_54_9536.log)

StoneLiBambu, 2024-07-26:
Hi @spawnofbill @rekonnected ,
    Thank you for the problem you reported. We tried to find a way to solve this problem, which did not adapt well to the login problem on Apple platform. We are very sorry.

Thanks!
Stone Li

Olli-M, 2025-10-20:
It's a problem when connecting to localhost in order to transfer the login details to bambu studio

rekonnected, 2025-10-20:
Good catch, I missed that coming back to this issue. Good that we have a workaround now. Still, having to disable a standard security feature to get a login function working isn't great and still leaves man-in-the-middle risks.

Seems like the next question to figure out this issue's source is why is localhost looped in how it is, and can that flow be changed to use valid addresses/domains throughout?