bambulab/BambuStudio #4290: 无法导出全部材料预设

yueduz, 2024-06-15:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

archlinux

### Additional system information

# 系统详情报告
---

## 报告详情
- **生成日期：**                                        2024-06-15 10:30:49

## 硬件信息：
- **硬件型号：**                                        ASRock B550M Phantom Gaming 4
- **内存：**                                          16.0 GiB
- **处理器：**                                         AMD Ryzen™ 5 5600 × 12
- **显卡：**                                          AMD Radeon™ RX 6600
- **磁盘容量：**                                        304.1 GB

## 软件信息：
- **固件版本：**                                        P2.90
- **操作系统名称：**                                      Arch Linux
- **操作系统内部版本：**                                    rolling
- **操作系统类型：**                                      64 位
- **GNOME 版本：**                                    46
- **窗口系统：**                                        Wayland
- **内核版本：**                                        Linux 6.9.4-arch1-1

### Printer

Bambu A1

### How to reproduce

导出预设包

### Actual results

[image]
有天瑞的预设，但在导出列表里不可见

### Expected results

导出所有预设

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15845390/log.zip)
[Project.zip](https://github.com/user-attachments/files/15845394/Project.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

DanBao-Bambu, 2024-06-19:
Can you package and send me the preset "天瑞" for analysis? My email is maosheng.wei@bambulab.com. It is located in the user folder at the same level as the log folder and can be searched or directly packaged and sent.

DanBao-Bambu, 2024-06-26:
This issue is due to the "compatible_printers" field being empty in the custom Filament you created, and it will be fixed.