bambulab/BambuStudio #4292: 3MF files exported from Bambu Studio are not made to specification

gjarnos, 2024-06-15:
### Bambu Studio Version

1.9.2

### Where is the application from?

Bambu Lab Official website

### OS version

All

### Printer

Not applicable

### How to reproduce

Export a 3MF

### Actual results

3MF files exported from Bambu Studio are not made to specification 

### Expected results

I've noticed `paint_color` is not part of the 3MF specification although it is in the 3MF's exported from Bambu Studio only. Can the team at Bambu please outline all of the different ways that Bambu has diverged from the 3MF specification so that proper parsers can be developed. Thank you.

### Project file & Debug log uploads

Not Applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

lanewei120, 2024-06-17:
the 3mf exported from Bambu Studio are made to specification
the 'paint_color' is just "other attrubute"
[image]

you can skip it

for Bambu's 3mf, you can refer to https://wiki.bambulab.com/en/software/bambu-studio/3mf-compatibility for the details

if you want to parse this 3mf, you can just refer to the codes in bambustudio

gjarnos, 2024-06-17:
Hi @lanewei120. Thank you for responding. I have another question regarding `paint_color`. Why not use the base material or color group which is fully supported by lib3mf?

lanewei120, 2024-06-17:
that color is used for the original trangles or group
our paint_color is derived from prusa slicer, which can be  used for the triangles splitted from the original

gjarnos, 2024-06-17:
When you say it's derived from Prusa Slicer, are you saying they use `paint_color`?

lanewei120, 2024-06-17:
they use a different name
[image]

lanewei120, 2024-06-26:
not a issue,  close it

closed 2024-06-26 by lanewei120

t-paul, 2024-10-12:
My reading of that part of the specification

> Within this core XSD schema (see Appendix B.1. 3MF XSD Schema), extension points have been explicitly entered in the
form of <any> elements and <anyAttribute> (also visible in the element diagrams further along in this specification). These
are required to come from other namespaces, which SHOULD point to a way to find the appropriate specification and
accompanying XSD schema.
> Vendors MIGHT define private 3MF extensions. The specifications of private namespaces (i.e. that are not ratified by the
3MF Consortium) need to be negotiated between parties in the ecosystem.

is that the naming shown in https://github.com/bambulab/BambuStudio/issues/4292#issuecomment-2172079128 is spec compliant as it's using a separate namespace. Only using `paint_color` from the core namespace seems to violate the requirement given in that paragraph.

I'm just noting that while implementing the color import for OpenSCAD. We will not be able to support that kind of coloring as we use the official 3MF library which seems to have no means of reading non-spec data (at least in library version v1, will be looking at v2 next).

It would be great finding a solution that is covered by official specification.

t-paul, 2025-04-23:
https://github.com/prusa3d/PrusaSlicer/issues/12502#issuecomment-2037338782 seems to agree with my earlier statement.

Current state now is:
* Multiple slicers use different attributes, but each needs to try reading the others too
* BambuStudio uses a non-spec compliant one, PrusaSlicer uses a technically spec compliant one (using different namespace)
* None of the ways are supported by the official 3MF library

I would not call this a good and future proof solution. Anyone who has contacts to the 3MF foundation to find an official and unified way of specifying this information?

t-paul, 2025-04-24:
> As a potential option, if you can get the lib3MF data out of the prusa or Bambu files

That is exactly the problem. The official [3mf library](https://github.com/3MFConsortium/lib3mf), even in the latest version has no means to access that data at all (see also your own still unanswered question https://github.com/3MFConsortium/lib3mf/issues/377).

OpenSCAD is a very small projects in regard of development power, It's not feasible to add and maintain a custom XML parser where a fully supported and officially maintained library exists (especially with the very much non-trivial internal structure of 3MF files).

FastFrench, 2025-04-25:
I've noticed that current OrcaSlicer handles colors a different way in .3mf files. 
This line in the xml models let me think that this behavious is or was somewhat related to bambulab
[image]

Even when the model is colorized (by filaments assignments), the colors do not appear in any way:
[image]

And here in fact, the color information only relies on metadata, where an extruder is associated to each object, and then a color defined to each extruder/ 
[image]
(in model_settings.config)

[image]
(in project_settings.config)

This is quite an ugly way to code colors, isn't it?
Does or did Bambu Studio sometimes code colors that way?

t-paul, 2025-04-27:
Well, we are talking on a closed issue here, so it's a bit like people shouting at clouds. I wonder what the best option is to bring that to some actual attention.

Considering Prusa is member of the 3MF Consortium (or was?, I can't find any reference on the current member page), maybe a new issue there could get things moving?

Ultimaker *is* still listed as member, but I don't know what the state of Cura is in that regard, and if they would be interested in driving some sort of specification effort.