bambulab/BambuStudio #4295: Updated access code is not remembered

CFenner, 2024-06-15:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

MacOS 14.5

### Printer

X1 Carbon

### How to reproduce

- pair a X1 Carbon to the app
- update the access code on the printer 
- restart the app & access the printer
- app requests the updated access code, connection works
- restart the app
- select devices tab, printer is listed as know device
[image]
- select printer, error "wrong password"
[image]
- printer I listed as unknown device
[image]
- select device & provide new access code, everything works as usual

This I need to repeat after each disconnect. So it seams that the app is not remembering the updated access code.

### Actual results

Printer is not connected and app requests a new pairing code.

### Expected results

Printer access is remembered.

### Project file & Debug log uploads

[Archiv.zip](https://github.com/user-attachments/files/15846282/Archiv.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vanyolai, 2024-06-15:
I have the same issue on Linux.

icebalm, 2024-06-16:
Same issue here, 1.9.2.57 started randomly forgetting my A1 Mini's access code and is making me re-enter it, also on Linux.

walterwongbbl, 2024-06-28:
Hi @CFenner 
Was everything normal in the previous version?

CFenner, 2024-06-28:
I just got my printer, no idea.

icebalm, 2024-06-28:
> Hi @CFenner Was everything normal in the previous version?

For me, everything was normal in the version previous to 1.9.2.57.

Rello, 2024-06-28:
+1 on 1.9.3.50

asking for the access code every time when restarting Bambu Studio

JulianGro, 2024-06-28:
For me, it actually stopped doing this, so this might be harder to figure out that in seems.

When I first got the printer last week, Bambu Studio would always require me to put in the access code again after power cycling the printer. I used the Bambu cloud for a couple of days and then switched back to LAN mode, and now it just works. One thing that changed is that I put the Wi-Fi settings onto the SD-Card instead of putting them in via the app. There was no firmware update and I didn't update Bambu Studio.

KizzyCode, 2024-06-29:
Having the same problem. Printer P1S in LAN mode, Bambu Studio 1.9.3.50 on macOS.
Bambu Studio regularly disconnects for an unknown reason (my WiFi is pretty stable, and I've double-checked the logs to ensure it's not caused by a WiFi/IP connectivity loss on either side), and every time it disconnects it forgets the access code.

walterwongbbl, 2024-07-09:
It seems that there are some issues with synchronizing the access code, which makes it difficult to remember locally. We are trying to reproduce this issue.

Oscilllator, 2024-07-17:
+1 to this issue. I noticed that if I a second item within like an hour of printing the first item I don't have to enter the code. But any kind of long wait results in me needing to enter the code again.

Dorenix, 2024-07-25:
Same issue, connected the printer in Lan only, changed code, and now I need to retype it on restart every time.

diver-ch, 2024-08-14:
For me this solved it: https://github.com/bambulab/BambuStudio/issues/3481

KizzyCode, 2024-08-14:
Seems to work at first glance!

herrdeh, 2024-11-01:
A clean installed v 1.10.0.81 on ubuntu Linux has got the same issue.

jonidesousa, 2024-12-10:
same here on macos

isrplo, 2025-01-15:
Smae here. Macos.