bambulab/BambuStudio #4297: Fedora AppImage does not open when Exectuable as Program is enabled.

gomeazy, 2024-06-15:
### Bambu Studio Version

1.9.2

### Where is the application from?

Bambu Lab github releases

### OS version

Fedora Linux 40 (Workstation Edition) x86_64

### Additional system information

OS: Fedora Linux 40 (Workstation Edition) x86_64
Kernel: Linux 6.8.11-300.fc40.x86_64
Shell: bash 5.2.26
DE: Gnome 46.2
CPU: Intel(R) Xeon(R) E5-2670
GPU: AMD Radeon RX 590

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Download Fedora .AppImage from https://github.com/bambulab/BambuStudio/releases/tag/v01.09.02.57.
2. Move downloaded .AppImage file to desired location on computer.
3. Right-click .AppImage and enable Executable as Program.
4. Double-click .AppImage.

### Actual results

The program does not launch or open. No error is displayed.

### Expected results

BambuStudio should open.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bambu123, 2024-06-17:
Is there any other way to run it?
I'm not sure what "Executable as Program" did. You can start the program in the console and see what logs are output and if there are any crashes.

artsai, 2024-06-19:
I just got the Bambu printer today. 
I had the same issue.  But when running the AppImage file in the terminal per @bambu123's suggestion, I can see this error. 
`bash
error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory
`

And when I looked up online it seems like you need to install mesa-libOSMesa
`bash
sudo dnf install mesa-libOSMesa
`
Then it will open.

gomeazy, 2024-06-19:
Thanks @artsai and @bambu123 for your input. @bambu123 like @artsai , I too was missing some libraries. When I tried to run the .AppImage from terminal, I received the following error: `error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory`.

When I checked my system with `dnf list webkit2gtk*` I could see that I had `webkit2gtk4.1.x86_64` installed. I couldn't figure out how to _only_ install `libwebkit2gtk-4.0.so.37` so I searched the error on Google and found that I was able to resolve this error with `sudo dnf install webkit2gtk4.0-devel`.

After installing `webkit2gtk4.0-devel` I was then presented with the same error that @artsai received: `error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory`. I followed @artsai's instruction by using `sudo dnf install mesa-libOSMesa`.

When I launch the .AppImage now, I get _use system SSL certificate: /etc/pki/tls/certs/ca-bundle.crt
To manually specify the system certificate store, set the SSL_CERT_FILE environment variable to the correct CA and
restart the application_ which I am familiar with with older versions of the .AppImage.

This case can now be closed however there should be some better instruction to let people know what they should have installed ahead of trying to launch the .AppImage. I only say this because not all Linux users are comfortable with Terminal.

MackBambu, 2024-07-10:
To run appimage on fedora 40 you need to:
```
sudo dnf install webkit2gtk4.0 libOSMesa
sudo dnf update
```
Appimage for fedora40 and ubuntu24 will be released separately soon.

loomsen, 2024-07-26:
Unfortuantely, this is still not working for me. I was just about to place my order for a printer, when I thought let's look at the software first.

```
$ rpm -qi mesa-libOSMesa
Name        : mesa-libOSMesa
Version     : 24.1.4
Release     : 3.fc40
Architecture: i686
Install Date: Fr 26 Jul 2024 11:36:26 CEST
Group       : Unspecified
Size        : 13813600
License     : MIT AND BSD-3-Clause AND SGI-B-2.0
Signature   : RSA/SHA256, Mo 22 Jul 2024 17:47:48 CEST, Key ID 0727707ea15b79cc
Source RPM  : mesa-24.1.4-3.fc40.src.rpm
Build Date  : Mo 22 Jul 2024 16:24:29 CEST
Build Host  : buildhw-x86-05.iad2.fedoraproject.org
Packager    : Fedora Project
Vendor      : Fedora Project
URL         : http://www.mesa3d.org
Bug URL     : https://bugz.fedoraproject.org/mesa
Summary     : Mesa offscreen rendering libraries
Description :
Mesa offscreen rendering libraries.

$ rpm -ql mesa-libOSMesa
/usr/lib/.build-id
/usr/lib/.build-id/03
/usr/lib/.build-id/03/365e69c327decbffe34387810282f3765b2dd5
/usr/lib/libOSMesa.so.8
/usr/lib/libOSMesa.so.8.0.0

$ ./Bambu_Studio_linux_fedora-v01.09.03.50.AppImage 
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
QSocketNotifier: Can only be used with threads started with QThread
/tmp/.mount_Bambu_gnPHDw/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: cannot open shared object file: No such file or directory
```

It can't find it, even though the lib is there. If I naively link it to `/usr/lib64` it complains about being the wrong ELF (which it obviously is)

```
$ ./Bambu_Studio_linux_fedora-v01.09.03.50.AppImage 
/usr/bin/AppImageLauncher: /lib64/libcurl.so.4: no version information available (required by /usr/bin/../lib/x86_64-linux-gnu/appimagelauncher/libappimageupdate.so)
QSocketNotifier: Can only be used with threads started with QThread
/tmp/.mount_Bambu_Q0dIuX/bin/bambu-studio: error while loading shared libraries: libOSMesa.so.8: wrong ELF class: ELFCLASS32
```

frank-pian, 2024-07-26:
@loomsen 
I suggest you try the flathub version.
https://flathub.org/apps/com.bambulab.BambuStudio

loomsen, 2024-07-26:
@frank-pian Thank you, that one works indeed :+1: