bambulab/BambuStudio #4300: Nozzle clumping detection still performed when turned off

Shedalizzm, 2024-06-16:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10 Home

### Printer

Bambulab A1 Mini

### How to reproduce

When printing the "Nozzle clumping detection" move is done even if I have disabled it in "Print options" on the screen on the printer

### Actual results

After the first few layers, print head goes off to the side of the print bed and lowers down to check for clumping. This happens when its enabled but also when its disabled.

### Expected results

The nozzle clumping detection movement should not happen when its turned off.

### Project file & Debug log uploads

[Nozzleclumpingproblem.zip](https://github.com/user-attachments/files/15859434/Nozzleclumpingproblem.zip)
[bambulog.zip](https://github.com/user-attachments/files/15859436/bambulog.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

LiveDoBond, 2024-06-23:
Same problem here, it start after the new firmware.

LiveDoBond, 2024-06-24:
Solved the problem hear, at last i don't get the issue for now.

I solved removing ALL PTFE TUBEs and re-organizing one by one to do less curve possible.

And now problem appear to be solved.

XunZhangBambu, 2024-07-12:
@Shedalizzm After turning off the nozzle clumping detection, we will still perform the clumping detection at the third layer

sutr90, 2025-06-02:
@LiveDoBond You can disable the 3rd layer check in the printer's machine code setting. Open BambuStudio, edit the printer, and in the Machine Code section there is a set of instructions for Timelapse mode. 

If you delete this section, the 3rd layer clog detect will be disabled (along with timelapse, of course). Save it as new preset, and you are good to go.