bambulab/BambuStudio #4301: Build fails on Gentoo

desultory, 2024-06-16:
### Bambu Studio Version

BambuStudio-01.09.02.57

### Where is the application from?

Bambu Lab github releases

### OS version

Gentoo 23.0

### Printer

Bambu A1 Mini

### How to reproduce

extract the release tarball,
```
cd BambuStudio/deps
mkdir build;cd build
cmake ../ -DDESTDIR="/home/<user>/tmp/BambuStudio-01.09.02.57/BambuStudio_dep" -DCMAKE_BUILD_TYPE=Release -DDEP_WX_GTK3=1
make -j32
```

### Actual results

Dependencies fail to build.

### Expected results

Dependencies build.

### Project file & Debug log uploads

https://0x0.st/Xc5C.txt

### Checklist of files to include

- [X] Log file
- [ ] Project file

desultory, 2024-06-16:
[build.log](https://github.com/user-attachments/files/15859653/build.log)

JuergenKammer, 2024-06-20:
That is a failure building OCCT, thus /maybe/ a duplicate of issue #4158. I posted my solution there. Try changing the OCCT.cmake as I described there. (Had the error on debian linux, would guess its a general one).

desultory, 2024-06-20:
Thanks for the info, I'll try that.

Could these deps somehow be submodules or something? or could there be an option for system libraries to be used?
I feel this process could be improved a bit.

desultory, 2024-06-20:
That patch works, thanks.

lanewei120, 2024-06-26:
thanks, @MackBambu will fix it soon

closed 2024-08-26 by desultory