bambulab/BambuStudio #4303: Model painting dissappear after reloading geometry

Namiiria, 2024-06-17:
### Bambu Studio Version

1.9.1.67

### Where is the application from?

Bambu Lab Official website

### OS version

Windows 10

### Additional system information

Processor	12th Gen Intel(R) Core(TM) i5-12600K   3.70 GHz
Installed RAM	32.0 GB (31.8 GB usable)
System type	64-bit operating system, x64-based processor
RTX 3060 TI

### Printer

A1 Mini

### How to reproduce

1. paint a model
2. make a small minute change to the model (adjusting the position of one vertice)
3. reload from disk in bambu studio

[https://youtu.be/KcoAq5Mcz6I](https://youtu.be/KcoAq5Mcz6I)

### Actual results

The model's painting get fully reset

### Expected results

The model geometry should be updated and keep the painting to its best ability. Or at least let us export/import the painting data as Vertex color, OR FBX import/export with UV maps so we can save it as a texture.

Right now this issue makes the Bambustudio-based painting workflow very risky, if I decide to paint the model, then absolutely no changes ever can be made on it... otherwise I will have to repaint it from scratch. Considering the iterative nature of tinkering and modelling for 3Dprints, this is a strong problem for me, preventing me from accommodating feedback from downloaders of the model on makerworld (without spending 1-2 hours every time re-painting the same thing).

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15860309/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Haidiye00, 2024-06-17:
Hello, please record a GIF to explain your issue. Thank you

Namiiria, 2024-06-17:
[image]
here is it in gif form :)

Haidiye00, 2024-06-18:
May I ask if your original model is an stl model

closed 2025-04-16 by Haidiye00