bambulab/BambuStudio #4307: [Feature Request] – PID Tuning (and storing individual values)

SnickCH, 2024-06-17:
**Is your feature request related to a problem? Please describe.**
Now with the official PANDA Revo hotend that is supported, there is a need for PID tuning. Of course with all different after market hotend, this would be grat too. At the moment with some hotends the values are not correct and therefore it overshots with the heat.**Describe the solution you'd like**

**A clear and concise description of what you want to happen.**
1) PID Tuning in Bambustudio 
2) Safe the values as a preset (so we can change the values when we change the hotends)
3) Configure / Change the PID values manually (without running a PID tuning / test). 

**Describe alternatives you've considered**
There is no alternative to this. Only printing with "wrong" PID values.

OsmosisD, 2024-08-21:
Would also like this. The Panda comes with a 60W heater instead of matching the 48W stock heater, plus the thermistor location is different. I'm getting actual temps 10-15°C higher than the machine is reporting. You can also check out YGK3D's video where he did a test: https://youtu.be/dNmWJcONe5E?feature=shared&t=117

There's also the Haldis TZ2.0 and TZ3.0. Heck, even the official E3D Bambu HF/ObXidian nozzles will have different heating characteristics and an option for PID tuning/calibration would be useful for these.

Maunose, 2025-02-28:
I would love to see this implemented!