bambulab/BambuStudio #4311: Bambu Studio "Filament Sync" synchronises with Select Printer/AMS in Device Tab

darkneon2002, 2024-06-17:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Win 11 x64 Home

### Printer

X1C, P1S

### How to reproduce

Device setup:
Printer A connected to AMS AA
Printer B connected to AMS BB
Important: AMS AA & AMS BB are not connected to each other with an AMS Hub

Steps:
In Bambu Studio go to prepare and open a project or create a new project.
a blank project will do.

In Prepare: Select Printer A
in Device: Select Printer B
In Prepare: Press AMS Sync icon
Select Sync or Resync

### Actual results

Following the above steps results in that; Printer A will now be synced with the filaments in the AMS connected to Printer B

### Expected results

The expected result is that Printer A syncs with AMS AA regardless of the selected Printer B in the device tab since no AMS Hub is connecting AMS AA and AMS BB with Printer A

### Project file & Debug log uploads

[Shelf plates.zip](https://github.com/user-attachments/files/15870943/Shelf.plates.zip)
[Uploading log.zip…]()

### Checklist of files to include

- [X] Log file
- [X] Project file

MadMax389, 2024-06-18:
Doesn't work that way. You have to select the correct printer on the device screen. I have suggested in another post that we have a drop down near the sync button that allows us to select the AMS to sync without having to switch back to the device screen.

darkneon2002, 2024-06-18:
well it should be your suggestion would even be better, no mistake possible

DanBao-Bambu, 2024-06-28:
Yes, the printer that prepares the page does not determine which printer the Device page is connected to. Sync and ReSync operations only synchronize the material information of the AMS of the current Device printer.