bambulab/BambuStudio #4315: crashes when slicing

slql0011, 2024-06-18:
### Bambu Studio Version

1.9.2.57

### Where is the application from?

Bambu Lab Official website

### OS version

Windows11

### Additional system information

CPU: intel 13700k
Memory: 32G

### Printer

Bambu A1 Combo

### How to reproduce

just slice

### Actual results

software would crash when slicing prints

### Expected results

Complete slicing

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/15879991/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

ttt888zzz, 2024-06-20:
Could you please upload the source file

QingZhangBambu, 2024-06-24:
[#3318](https://github.com/bambulab/BambuStudio/issues/3318)
Hi, could you try these methods? sometimes the CPU may not be stable enough to support multithreading.

closed 2024-06-25 by slql0011